DLR-SC/tigl #125: Half model guide curves does not touch the symmetry plane

tobsto, 2015-04-17:
[image]

rainman110, 2015-06-02:
@tobsto Can you please verify, why the test case FuselageGuideCurve.tiglFuselageGuideCurve_CCPACSFuselageSegment is failing. I don't really understand the test. There seem to be some assumption included that are not commented.

tobsto, 2015-06-03:
The reason for the test failure is the assumption, that the fuselage guide curve with alpha=0 between two segments (which are symmetric w.r.t. the x-y plane) lies in the x-y plane. Despite the fact that the fuselage profile points are symmetric w.r.t. the x-y plane, the resulting profiles are not due to the new algorithm.

rainman110, 2015-06-10:
The issue should be fixed now.

commit cf9c5ad2 (2015-06-10): Improved the bspline algorithm for exact symmetry

The algorithm now build the bspline by interpolating
in both directions and then taking the average curve.

fixes #125
closed 2015-06-15 by rainman110