DLR-SC/tigl #132: Wrong orientation of patch faces in guide curve case

tobsto, 2015-05-07:
The patch faces resulting from MakePatches are not oriented well. The normal vectors should point "outwards". At the moment, they are pointing "inwards":
[image]

rainman110, 2015-05-08:
This is probably the same bug as mentioned earlier. It is actually not possible to detect, what should be correct if we don't have a solid. In case of a solid it is easier, as we simply could take a point at infinity and check if it is outside (it should be), or not. In the latter case, we just have to invert the shell.

I think, the direction of the patches depends only on the directions of the guides and profiles. So we just have to provide guides and profiles with the "correct" direction.

tobsto, 2015-05-13:
I agree. To provide guides and profiles that yield the correct orientation is the cleanest way. However, in the end all surfaces will be closed, right? Then one could use the solid for detecting the orientation.

rainman110, 2015-05-13:
I am not entirely shure, if we'll get a solid in all cases. E.g. if we provide a entry and exit plane of the engine, the nacelle might not be a solid.

tobsto, 2015-05-13:
The normal of a surface patch is determined by a right handed coordinate
system attached to the (u,v) = (0,0) point of the surface patch. Here
the tangent of the profile defines the x-direction and the tangent of
the guide defines the y-direction. The normal is the z direction.

As a result, the resulting shell is correctly orientated if

- The direction of rotation of the wing profiles is the y-direction, and
  the wing guides run from the fuselage to the wing tip (in y-direction)

[image]

- The direction of rotation of the first (upper) nacelle profiles is in
  y-direction and the direction of rotation of the nacelle guides is in negative
  x-direction.

[image]
[image]

- The direction of rotation of the fuselage profiles is negative x-direction, and
  the fuselage guides run in x-direction (or vice versa)

commit daeeb78a (2015-05-13): Added test with correctly orientated nacelle guides

The normal of a surface patch is determined by a right handed coordinate
system attached to the (u,v) = (0,0) point of the surface patch. Here
the tangent of the profile defines the x-direction and the tangent of
the guide defines the y-direction. The normal is the z direction.

As a result, the resulting shell is correctly orientated if

- The direction of rotation of the wing profiles is the y-direction, and
  the wing guides run from the fuselage to the wing tip (in y-direction)

- The direction of rotation of the fuselage profiles is neg [... cut]
closed 2015-05-13 by tobsto