DLR-SC/tigl #145: Guide curve lofting algo crash since changes

rainman110, 2015-05-27:
Since commit 60eff72bc8c0 (sequential fusing), the lofting algo crashes in case of the heli body (ROBIN_mod7.cpacs.xml).

rainman110, 2015-05-27:
Another problem arised with the xrf1 fuselage loft. This time it is another commit, that introduces the error: d5413306160a0a3dd638571b4dff4ee873523128

This is important tp fix, since this will break the xrf1 config.

rainman110, 2015-05-27:
The problem with d541330 is a too small maxLoops variable. The fuselage requires 60 loops. We need a check that does not require any fixed number (if possible).

rainman110, 2015-05-27:
The number of required loops is identical to the number of generated faces. Thus, we can estimate the number of required loops beforehand:
  - if guides and profiles are open: loops = (nguides-1)*(nprofiles-1)
  - if guides are open, profiles closed: loops = (nguides)*(nprofiles-1)
  - if both guides and profiles closed: loops = nguides*nprofiles

rainman110, 2015-05-28:
The problem with the sequential fuse still exists...

rainman110, 2015-05-28:
The reason why sequential fuse was required, was an invalid input shape. The question is, __why the input shape (guide curves) were invalid__? The makepatches algorithm seems to be picky about the tolpar value. Only with a correct value (<1e-7) the resulting shapes are valid.

tobsto, 2015-08-21:
This may be fixed together with issue #170

rainman110, 2015-12-17:
@tobsto  Can we close this issue?

tobsto, 2015-12-18:
Yes. I think so.

closed 2015-12-18 by tobsto