DLR-SC/tigl #146: Too complex test case

rainman110, 2015-05-29:
The tests for the makeloops algorithm are too complicated and too lenghty. Also, there is an assertion that one algorithm __should__ fail. This makes no sense, as we only expect the algorithm to fail. Still, it does not matter if it wouldn't fail, due to some improvement of the algorithm.

This assertion makes one test fail, just because some improvement in code.

If we really want to test failures, we have to provide some artificial example, where it is clear, why the algorithm must fail.

rainman110, 2015-05-29:
Here's the test failure of travis: https://travis-ci.org/DLR-SC/tigl/builds/64482838

rainman110, 2015-06-10:
Test is still failing. @tobsto, can you please correct the test?

rainman110, 2015-07-31:
Don't want to annoy, but could you please fix the test case (@tobsto) ;)

tobsto, 2015-08-03:
I cleaned up the test case and remove the "should fail" assertion. The SetUp function is still lengthy. However, there is no alternative since we want to test the MakeLoops algorithm and we need the preparatory fusing of guides and profiles. The latter can not be pulled out of the MakePatches algorithm.

closed 2015-08-03 by tobsto