DLR-SC/tigl #170: Fix fusing of surfaces built along guide curves

tobsto, 2015-08-10:
Provide a robust way to fuse surfaces which were built by the loft maker along guide curves.

tobsto, 2015-08-21:
Introduced test in 9bfa4ccfae4186e19f8eb1bea29c34cc9ba42cd3 which test the correct fusing of the wing and the fuselage. It will fail for unpatched OCC versions. Now we need to set up and test the whole chain to make sure no other bugs were introduced by the fusing operation.

1.) Creating and exporting the fused geometry with TIGL like in the example script at https://gitlab.sc.dlr.de/ilic_ca/tiglsurf

2.) Test maplib forward and backward for a whole mesh on the tigl geometry (create e.g. with Salome/Netgen). So far this is failing due to a maplib issue: https://gitlab.sc.dlr.de/tigl/maplib/issues/15

rainman110, 2015-10-21:
With commit 858c057 all tests are running.

tobsto, 2015-10-23:
After adding an extensive parameter study to the confidential tests, 2 testcases fail: 

https://gitlab.sc.dlr.de/TIGLConf/tigl-confidential-big-test-data/blob/679593e99e48420f8ffaf7c243c1a14753cf1310/dx-xrf1-wfhv-wing-change/cpacs07.xml

and 

https://gitlab.sc.dlr.de/TIGLConf/tigl-confidential-big-test-data/blob/679593e99e48420f8ffaf7c243c1a14753cf1310/dx-xrf1-wfhv-wing-change/cpacs52.xml

Maybe the first one could be neglected since the wing intersects the symmetry plane. However the second one does not. Further investigation is needed.

tobsto, 2015-11-06:
In the context of #182 we implemented a new parameter study in e7d9f6aafdbf07b5177999049992416e93cab546 which avoids the intersection of the wings with the symmetry plane. See #182 for details. All relevant tests for fusing and mesh generation with Centaur are successful. Therefore the issue is closed.

However, the above mentioned failed test case may indicate that future problems with the fusing can not be ruled out. Keep in mind, that we fixed this problem by a workaround.

closed 2015-11-06 by tobsto