DLR-SC/tigl #171: Fused geometry of mirrored half model is not mirror symmetric

tobsto, 2015-08-19:
The result of an aircraft fuse algorithm for a mirror symmetric half model is not mirror symmetric: The wing is still inside the fuselage for one side:

[image]

Reproduce with:
```
tigl::PTiglFusePlane fuser = config.AircraftFusingAlgo();
fuser->SetResultMode(tigl::FULL_PLANE);
PNamedShape airplane = fuser->FusedPlane();
```

rainman110, 2015-09-01:
I cannot reproduce the bug. I created a mirror symmetric version of the XRF1 using TiGL Viewer with occt 6.7.0 (oce-0.17) and with occt 6.9.0. The screenshot is the following:
[image]

Please provide more information (which model, which occt version) to be able to reproduce this error.

rainman110, 2015-09-01:
I could now reproduce the bug with the latest xrf1 data from caslav. Fortunately, its not really a bug. As a temporary hack for the fusing problems, we introduced a small shift of 0.05m of the fuselage. This lead to a self intersection of both fuselage halves. If we remove this hack, everything works as expected.

closed 2015-09-01 by rainman110