DLR-SC/tigl #176: Component segment eta,xsi in boxwing

robrousseau, 2015-10-22:
I have attached two screenshots that show the location of three (eta,xsi) coordinates on a boxwing components segment. The entire boxwing contains one component segment. The first point (0,0) is located at the root leading edge, as expected. The second point (0.5,0) is located at the leading edge of the middle of the connection part, which was expected as well. The third point however (1,0) is not located at the last segment leading edge. It seems to be floating too far inboard, with an incorrect x,y and z coordinate. This problem only occurs for boxwing aircraft, when the wing segments start moving inboard again.

[image]
[image]

rainman110, 2015-10-22:
Thanks for the report. This is indeed a bug. Could you please send me the CPACS file for further analysis?

robrousseau, 2015-10-22:
Dear Mr. Siggel,

Thanks for your quick reply.
We have noticed that the issue arose on TIGL viewer 2.1.3, but does not occur in TIGL viewer 2.1.6.
I attached the xml file.

Best regards,

Rob Rousseau

rainman110, 2015-10-22:
Unfortunately, there is not cpacs file attached. Can you please mail it to me (martin __dot__ siggel __at__ dlr __dot__ de).

I suspect also errors with new TiGL versions. The errors might be however different than the older versions, since the code for the function has changed.

commit 93fdaf34 (2015-10-23): Fixed an error in CCPACSWingComponentSegment::GetPoint

The error occurred in case of multiple intersections of the eta plane
with the wing. This is the case for e.g. box wings. The current solution
computes all intersection points, projects them to the local y-z plane of
the wing and computes the distance for each point to the eta point.
The point with the smallest distance is returned.

Fixes #176
closed 2015-10-23 by rainman110