DLR-SC/tigl #182: Disconnected Panels in Centaur after IGES import

tobsto, 2015-10-26:
Centaur complains about disconnected panels after import of IGES files create by TiGL:
[image]

tobsto, 2015-10-30:
All the [big testcases in conftests](https://gitlab.sc.dlr.de/TIGLConf/tigl-confidential-big-test-data/tree/679593e99e48420f8ffaf7c243c1a14753cf1310) seem to work with f3a462d2c1d74b24ec3d43fe5f4f735c18829ff8 except the parameter sweeps we generated: [1](https://gitlab.sc.dlr.de/TIGLConf/tigl-confidential-big-test-data/tree/679593e99e48420f8ffaf7c243c1a14753cf1310/dx-xrf1-wfhv-horiztail-change) [2](https://gitlab.sc.dlr.de/TIGLConf/tigl-confidential-big-test-data/tree 679593e99e48420f8ffaf7c243c1a14753cf1310/dx-xrf1-wfhv-verttail-change) [3](https://gitlab.sc.dlr.de/TIGLConf/tigl-confidential-big-test-data/tree/679593e99e48420f8ffaf7c243c1a14753cf1310/dx-xrf1-wfhv-wing-change)

Here, it turns out the base geometry had some issues concerning the intersection of the wing, the fin and the tailplane with the symmetry plane:
[image]

We need to fix the base geometry and regenerate the parameter sweep. Here, we need to make sure, that no intersection of wings with the symmetry plane takes place.

tobsto, 2015-10-30:
In order to make the base geometry work, we move the wing and the tailplane away from the symmetry plane and check if the resulting IGES export is valid in Centaur by [this script](https://gitlab.sc.dlr.de/TIGLConf/tigl-confidential-tests/blob/a5d5f69c105d7f27ddf3251aa89b213348c5fed2/centaurIGESTest.py). We also shift the fin slightly over the symmetry plane for the same reason. We tested to following translations:

 Wing translation in y-direction   |   Valid IGES in Centaur
----------------------------------------|----------------------------------
 0 | Yes
1E-6 | Yes
1E-5   | Yes
1E-4 | Yes
1E-3 | Yes
2E-3 | Yes

 Tailplane translation in y-direction   |   Valid IGES in Centaur
----------------------------------------|----------------------------------
 0 | Yes
1E-6 | No
1E-5   | Yes
1E-4 | Yes
1E-3 | Yes
2E-3 | Yes

 Fin translation in y-direction   |   Valid IGES in Centaur
----------------------------------------|----------------------------------
 0 | No
-1E-6 | Yes
-2E-6   | Yes
-5E-6 | No
-1E-5 | No
-2E-5 | No
-5E-5 | No
-1E-4 | Yes
-2E-4 | Yes
-5E-4 | Yes
-1E-3 | Yes
-2E-3 | Yes
-5E-3 | Yes
-1E-2 | Yes

Therefore, we choose the base configuration to have the following CPACS parameter:

Wing y-translation | Tailplane y-translation | Fin y-translation
-----------------------|------------------------------|----------------------
1E-6 | 1E-4 | -1E-4

The corresponding CPACS file is [here](https://gitlab.sc.dlr.de/TIGLConf/tigl-confidential-tests/blob/6f198252d272bd51af07e95022c0ecdedbb377d2/TestData/xrf1_parameter_sweep_base.xml).

tobsto, 2015-10-30:
Setting up new parameter sweep with 7bfd47343ea15986a8a18b12536825412adb8452. The corresponding parameter ranges can be found in the configuration files for the [sweepCPACS](https://gitlab.sc.dlr.de/tigl/sweepCPACS) tool:

1. [The translation and rotation of the wing](https://gitlab.sc.dlr.de/TIGLConf/tigl-confidential-big-test-data/blob/d0b486fb31e176a5d6f5f7b2a968ba4a448e4025/dx-xrf1-wfhv-wing-change/sweepcpacs.config)
2. [The translation and rotation of the tailplane](https://gitlab.sc.dlr.de/TIGLConf/tigl-confidential-big-test-data/blob/d0b486fb31e176a5d6f5f7b2a968ba4a448e4025/dx-xrf1-wfhv-horiztail-change/sweepcpacs.config)
3. [The translation and rotation of the fin](https://gitlab.sc.dlr.de/TIGLConf/tigl-confidential-big-test-data/blob/d0b486fb31e176a5d6f5f7b2a968ba4a448e4025/dx-xrf1-wfhv-verttail-change/sweepcpacs.config)

rainman110, 2015-10-31:
Great analysis. But now you should take some time off! Thanks!

tobsto, 2015-11-06:
Despite the above analysis. The minimal distance to the symmetry plane seems not to bee enough: Some tests with the Centaur [script]https://gitlab.sc.dlr.de/TIGLConf/tigl-confidential-tests/blob/a5d5f69c105d7f27ddf3251aa89b213348c5fed2/centaurIGESTest.py) still fail. In order to be on the save side, we choose the base configuration to have the following CPACS parameter:

Wing y-translation | Tailplane y-translation | Fin y-translation
-----------------------|------------------------------|----------------------
1E-4 | 1E-4 | -1E-3

Also we need to couple the rotation of the wing around the x-axis with a large translation in the y-direction. Otherwise an intersection with the symmetry plan occurs. 

This was done with 58930d4cc4cfbe803e81725841fd1a851f53359e. Also we fixed an accidental confusion of parameters here. The corresponding parameter ranges can be found in the configuration files for the [sweepCPACS](https://gitlab.sc.dlr.de/tigl/sweepCPACS) tool:

1. [The translation and rotation (y-axis) of the wing](https://gitlab.sc.dlr.de/TIGLConf/tigl-confidential-big-test-data/blob/88dc0433b7b86477d26bf3725a6129c490658f3edx-xrf1-wfhv-wing-change/sweepcpacs.config)
2. [The translation and rotation (x-axis) of the wing](https://gitlab.sc.dlr.de/TIGLConf/tigl-confidential-big-test-data/blob/88dc0433b7b86477d26bf3725a6129c490658f3e/dx-xrf1-wfhv-wing-xrotation-change/sweepcpacs.config)
3. [The translation and rotation of the tailplane](https://gitlab.sc.dlr.de/TIGLConf/tigl-confidential-big-test-data/blob/88dc0433b7b86477d26bf3725a6129c490658f3e/dx-xrf1-wfhv-horiztail-change/sweepcpacs.config)
4. [The translation and rotation of the fin](https://gitlab.sc.dlr.de/TIGLConf/tigl-confidential-big-test-data/blob/88dc0433b7b86477d26bf3725a6129c490658f3e/dx-xrf1-wfhv-verttail-change/sweepcpacs.config)

Since all [automatic tests] and the [Centaur test script] is successful, this issue is closed with 58930d4cc4cfbe803e81725841fd1a851f53359e.

closed 2015-11-06 by tobsto