DLR-SC/tigl #185: BUG: Vertical tail span with tiglwrapper 'wingGetSpan'

Luque9, 2015-11-22:
About CPACS file 'D150_AGILE' this is the script: 
---------------------
import tiglwrapper
tigl_h=tiglwrapper.Tigl()
vertical_tail_span = tigl_s.wingGetSpan("D150_VAMP_SL1")
--------------
Result is vertical_tail_span=7.11m instead 5.89m (value obteined from ‘.iges’ file opened in CATIA)

solutions?
Thanks

rainman110, 2015-11-23:
Can you please provide the cpacs file?

Luque9, 2015-11-25:
Thanks for you availability.

Luca Stingo
Date: Mon, 23 Nov 2015 12:42:47 -0800

  
  

	
		
			
		
		Questa e-mail è stata inviata da un computer privo di virus protetto da Avast. www.avast.com

rainman110, 2015-11-29:
By just looking at the chord points at (0,1) and (1,0), I get a span of 5.88. More investigation is required.

rainman110, 2015-11-29:
It seems, that the span is evaluated in x direction instead of z direction. The x-span is given as 7.11 m.

Luque9, 2015-12-03:
So, Is there no solution?

rainman110, 2015-12-03:
Let my state the true problem behind it: The span of the wing is not really properly defined. For example for the main wing, it might be the span in y direction, where for the VTP it is usually the z direction. 

I TiGL /CPACS, we have no semantics behind the wing object, so I have no idea, what kind of wing this is. This makes is very difficult to decide, in which main direction  the span should be evaluated.

What TiGL currently does is, that it determines the direction of the largest extend. For the main wing, this is usually the y direction. Fot the VTP, one would think, that this must be the z direction. In fact, it is the X direction. Therefore we get a wrong result.

Do you have a suggestion, how we could handle this in a better way?

Luque9, 2015-12-17:
@rainman110 the CPACS file contains x,y,z coordinates value of vertical tail root and the matching in CAD software is ok!

Instead about x,y,z coordinates value of vertical tail tip i didn't found corresponding values in CAD softaware. However if these values

are correct , is possible to evaluate vertical tail span like distance between z-coordinate tip and z-coordinate root (abs(z_tip - z_root))??

Let me know about this idea.

P.S. : what do you mean when you say span in X-direction?
 
Than you so much.

closed 2015-12-17 by rainman110

rainman110, 2015-12-17:
@Luque9  I fixed the bug. You can download a nightly build from http://sourceforge.net/projects/tigl/files/ . The bugfix will be included in the nightlies from tomorrow on.

commit b50b51bb (2015-12-17): Fixed bug in wing span computation

As we have no semantics in CPACS, what kind of wing (HTP, VTP ...)
we have, it is hard to identify the direction of evaluation for
the span computation. Until now, we just use the direction of the
wing whose bounding box is largest. That could produce errors, if
the depth of the wing is larger than its span. With the current
implementation, we ignore the depth by just looking at the leading
edge. We now choose the "major" direction of the leading edge.

Fixes #185