DLR-SC/tigl #190: Standard libraries should be searched for by cmake

sdeinert, 2015-12-16:
Under Linux, cmake should look for the standard libraries before the actual build process
/usr/lib64/libcurl.so
/usr/lib64/libxslt.so
/usr/lib64/libxml2.so
This way, if not found, specific paths to the libraries could be provided to cmake.
Currently, only make throws an error, if the libraries are not found under the standard folder and name.

rainman110, 2015-12-17:
TiGL does not explicitly link to these libs. These are dependencies of TiXI.

Did you download some binary package of TiXI or did you compile TiXI by yourself?

sdeinert, 2015-12-18:
We compiles Tixi ourselves. However, cmake does include these dependencies for e.g. the TIGLViewer.

Therefore, the error occurs when trying to link the TIGLViewer, even though the Tixi libraries are already compiled and available.

rainman110, 2015-12-18:
Hm... As I said. There is no explicit linking to these mentioned libraries, neither in TiGL nor in TiGL Viewer.

The only reason, It might try to link these libs could be, that TiXI exports these libs into ```INSTALL_PREFIX/lib/tixi/tixi-targets-release.cmake```. This can only make problems, if you compiled tixi on a different machine and copied it over to your machine. 

As a temporary workwround, you could remove all curl, libxml, libxslt entries from ```INSTALL_PREFIX/lib/tixi/tixi-targets-release.cmake```.

rainman110, 2016-01-05:
@sdeinert Did you fix your problems? Can we close this issue?

closed 2016-01-15 by rainman110