DLR-SC/tigl #195: function tiglwrapper WingGetSpan for the vertical tail (BUG)

Luque9, 2016-01-29:
the function of tiglwrapper (2.1.6 version, the last) gives a wrong value (8.14m) about the vertical tail span (the real value is 4.4m).
If you need the CPACS file i can send you an email.

rainman110, 2016-01-30:
This is probably the same issue as #185, which was already fixed in the Nightly builds.

We have not released a new TiGL version, therefore 2.1.6 still has this bug.

Please download a nightly build from http://sourceforge.net/projects/tigl/files/?source=navbar and test, if the bug is fixed in this particular case.

rainman110, 2016-01-30:
Duplicate to #185.

rainman110, 2016-02-10:
This bug still exists. In this case, the depth of the leading edge is larger than the span. This leads to the false assumption, that the wing extends in x Direction.

In order to improve this, we have to take the orientation of the sections into account. This will give us a depth coordinate which can be then excluded in the decision for the wing span direction.

commit 6656f026 (2016-02-15): Improved determination of the wing span direction

The new heuristic also takes the depth direction of
the wing into account and ignores this direction on
wing span evaluation.

Fixes #195
closed 2016-02-15 by rainman110