DLR-SC/tigl #198: Fuselage surface calculation fails

roadrunner2100, 2016-03-11:
Fuselage surface calculation fails (TIGL_ERROR (1)) for two different rotorcraft configurations. As alternative, trying to summarize over the segment surface areas, for all segment areas the value 0.0 is calculated. TIGL version used is TIGL2.1.6

[Bo105_v0_04_maserati.txt](https://github.com/DLR-SC/tigl/files/169538/Bo105_v0_04_maserati.txt)

rainman110, 2016-03-11:
This seems to be an opencascade problem.

The problem seems to be at least the following segments: Helicopter_Fuselage_struct_Seg_29_30 and Helicopter_Fuselage_struct_Seg_30_31

Fusing leads to a crash/exception of the algorithm. Further investigation required.

joergbrech, 2020-07-20:
@rainman110, I assume this can be closed then?

rainman110, 2020-07-20:
Yes, lets close it.