DLR-SC/tigl #200: Implementation of new ComponentSegment definition in CPACS

DLR-LY, 2016-03-29:
The ComponentSegment definition in CPACS will change soon. Since they are a main part of the wing definition the TIGL functions need to be changed accordingly in order to rovide an updated interface to all TIGL users.

rainman110, 2016-03-29:
We should discuss, how the change will be introduced. As this will break lots of user code, I suggest to release this change as a major CPACS release (i.e. CPACS 3).

Then, we could create a TiGL 3 release as well. The question is, how we will deal with old CPACS files:
 - discard them?
 - keep old api (+ deprecated old functions) and add new functions for the new definition

DroneDude123, 2016-06-30:
I would be interested in checking out the new definition. I'm considering using a subset of CPACS for a project of mine. Is there a place I can check it out?

DLR-LY, 2016-06-30:
We are still working on the new definition for the ComponentSegments in CPACS. A first CPACS version should be available at the end of this year. The TiGL developments can only start once the CPACS definition is fixed.
The CPACS developments can be followed on [https://github.com/DLR-LY/CPACS](url).

rainman110, 2017-05-29:
The new math for the GetPoint functions is implemented now as of 945a1c7aafaa1d989ded952d061628ab8c17aa83 and b6965421e4dace8049e5bfb47c11032e2916bd54.

What remains to be done is to use segment coordinates alternatively for positionings. This requires still changes to the cpacs schema, see https://github.com/DLR-LY/CPACS/issues/495.

I'll close this issue now. For the new feature, we'll open a new issue.

closed 2017-05-29 by rainman110