DLR-SC/tigl #203: CTiglPointTranslator terminates with TIGL_MATH_ERROR

RlanderRISCSW, 2016-08-11:
CTiglPointTranslator returns TIGL_MATH_ERROR in following example:

```
TEST(TiglPointTranslator, Bug5){
    CTiglPoint x1(4000.0000000000000, 0.00000000000000000, 3000.0000000000000);
    CTiglPoint x2(6893.7199999999993, 4250.0000000000000, 3000.0000000000000);
    CTiglPoint x3(7554.9955629914994, 0.00000000000000000, 3017.7749461533222);
    CTiglPoint x4(8693.7789747175666, 4250.0000000000000, 3006.3001935476932);

    CTiglPoint p(7060.6260619422628, 1062.5000000000005, 3011.1796935014363);

    double eta, xsi;

    CTiglPointTranslator trans(x1, x2, x3, x4);

    ASSERT_EQ ( TIGL_SUCCESS, trans.translate(p, &eta, &xsi) );
}
```

rainman110, 2016-08-28:
@RlanderRISCSW Thanks for the detailed report and test 😄 

I am gonna put this into our test suite. I suppose the reason could be the scaled number (factor 1000) and a hard-coded abort criterium.

rainman110, 2016-08-29:
Issue is fixed. The reason was indeed due to the different scale by a factor of thousand. That lead to the problem that the gradient of the objective function was 1e+6 larger than usual and the abort criterium was wrong.
I fixed the issue, by first determining the scale of the problem and adapt the tolerance of the gradient with the probem's scale.

closed 2016-08-29 by rainman110