DLR-SC/tigl #237: Change windows builds to VC 2015

rainman110, 2017-05-29:
As the build tools VC 2015 are now free to use inside the DLR, we should switch now our build tools to this platform.

This includes also generating all 3rd party dependencies to VC 2015 including:
 - [ ] OpenCASCADE / OCE
 - [ ] Freetype
 - [ ] Freeimage / FreeimagePlus
 - [ ] Qt (needs to be downloaded / is already shipped on appveyor)
 - [ ] TiXI
 - [ ] libxml2
 - [ ] libxslt
 - [ ] libcurl

rainman110, 2017-06-23:
The Jenkins builds should be switched not before TiGL Release 2.2.1.

joergbrech, 2017-07-28:
@rainman110 When trying to build the Matlab bindings with VC2015 I get an error because of these lines: 
https://github.com/DLR-SC/tigl/blob/9022b1c913bed88a1dd582d058f679088db652de/bindings/bindings_generator/matlab_cheader.in#L19-L24

Changing them to
```
/* Visual Studio C macros */
#ifdef _MSC_VER
  #define _CRT_SECURE_NO_WARNINGS
  #pragma warning(disable: 4090)
  #if ( _MSC_VER < 1900 )
    #define snprintf _snprintf
  #endif
#endif
```
according to this forum entry
https://stackoverflow.com/questions/27754492/vs-2015-compiling-cocos2d-x-3-3-error-fatal-error-c1189-error-macro-definiti
fixes the compiler error. Should I open a seperate issue and a PR for this?

rainman110, 2017-07-28:
Have a look into the Tixi repo and change it as it is done there:
https://github.com/DLR-SC/tixi/blob/master/bindings/bindings_generator/matlab_cheader.in

rainman110, 2017-09-03:
The release build pipeline works now completely on Travis and Appveyor. We don't use Jenkins anymore. Therefore, I close this issue.

closed 2017-09-03 by rainman110