DLR-SC/tigl #239: derive CTiglError from std::exception

dixxi, 2017-05-29:
TiGL currently throws CTiglError in erroneous scenarias, however, as CTiglError does not inherit from std::exception, it has to be explicitely caught in user code (or caught using an ellipsis in the catch clause).
Only in the first case an exception message can be obtained and reported, but this makes the client code depending on the CTiglError exception type.

Deriving CTiglError from std::exception would solve this issue. std::exception::what() could be overriden to return CTiglError::getError(), providing the error message via the standard std::exception interface.

I can prepare a PR for this change.

rainman110, 2017-05-29:
We should keep in mind, that CTiglError also carries the error code, which is important for the C API. 

@dixxi To get the error code, you have to explicitly catch CTiglError, right? 

Still, I think it is cleaner and more standard conforming to inherit from std::exception.

dixxi, 2017-05-30:
The catch clauses should look like this:
```
    catch (tigl::CTiglError& err) {
        LOG(ERROR) << err.what();
        return err.getCode();
    }
    catch (const std::exception& ex) {
        LOG(ERROR) << ex.what();
        return TIGL_ERROR;
    }
    catch (...) {
        LOG(ERROR) << "Unknown error in tiglProfileGetBSplineCount";
        return TIGL_ERROR;
    }
```
catching CTiglError first, returning the appropriate error code. std::exception is caught second, logging just the message, and ellipsis last.

PR #242 (2017-06-01): Cpacs 3 make tigl error exception
fixes #239
closed 2017-06-01 by rainman110