DLR-SC/tigl #258: Check accuracy of tiglWingGetSegmentUpperSurfaceAreaTrimmed

rainman110, 2017-06-24:
It seems, that ```tiglWingGetSegmentUpperSurfaceAreaTrimmed``` returns the same value for the upper wing skin and the lower wing skin. For asymetric profiles, this should not happen.

This must be checked!

joergbrech, 2017-07-06:
@CLiersch I cannot reproduce this error for D150 and D250. Could you provide a testcase where it fails? Thanks!

CLiersch, 2017-08-07:
@joergbrech: It seems, I cannot reproduce this any longer, either (in my current TiGL 2.2.0). Maybe I made a mistake when thinking to see this error, or it was already fixed by someone...

However, I had another look on the results and compared with summing up straight lines from Point to Point around the airfoil. For the D150 wing airfoil, the curve (or polyline, to be more precise) was roughly 3% longer than the chordline, and the lower side curve was roughly 0.3 % longer than the upper one. I could find these values also by using the abovementioned TiGL-functions, so it seems OK to me.

For the VTP, which is using a symmetrical airfoil, I further do get identical upper and lower curve lenghts from TiGL (this time around 5.5% longer than chordlength). This does also look good.

When looking at HTP (using same symmetrical. airfoil as VTP), I do not get the expected identical values for upper and lower side, but a relation of around 0.3% between each other. From my feeling, it looks a bit to much for being just an accuracy Problem (especially when comparing to the identical upper and lower VTP values). So it could be an indication that there is still something wrong. Please let me know if you need more details.
Best regards, Carsten

joergbrech, 2017-08-08:
@CLiersch , where exactly did you put the points on the wing to calculate the length of your polyline? 

When I calculate the lengths of *iso-eta* lines around a wing segment of the D150 I get symmetric results (i.e. same length on the lower and the upper part) both for the HTP and the VTP. I use the function `tiglWingGetLowerPoint` (resp. `tiglWingGetUpperPoint`) to calculate the points along the line and then I sum up straight line segments just like you did.

I do get a difference in the upper and lower surface area of about 0.36% for the HTP. But this is expected: The *iso-xsi* lines on the upper and lower part of the wing do not have the same lengths because of the non-zero `dihedralAngle` between the inner and the outer section. Setting the `dihedralAngle` to zero gives the same surface area for the upper and the lower wing segments.

Best regards, Jan

rainman110, 2017-08-08:
That totally makes sense to me. If this is really the case, the lower area should be larger than the upper one. @CLiersch if you're okay with the explanation we'll close this issue.

CLiersch, 2017-08-09:
You are absolutely right. Thanks for sorting this out. In the HTP case, it is the combination of sweep, dihedral and taper, whereas the VTP has only sweep and taper. Please close the issue.

closed 2017-08-09 by rainman110