DLR-SC/tigl #259: New Function to compute area of fuselage section

rainman110, 2017-06-24:
We need a function that computes the area of a fuselage section.

How to do:
 - Choose a fuselage segment
 - Get ISO-Curve of fuselage segment face (or cut with plane), defined by eta coordinate
 - Determine surface area of the resulting face

Proposed API:

    tiglFuselageGetSectionArea(handle, fuselageSegmentUID, eta)

CLiersch, 2017-06-26:
I would recommend to name the function
**tiglFuselageGetCrossSectionArea(handle, fuselageSegmentUID, eta, &area)**
because I think it is more intuitive easier to distinguish from other areas

rainman110, 2017-06-26:
> I would recommend to name the function
tiglFuselageGetCrossSectionArea(handle, fuselageSegmentUID, eta, &area)
because I think it is more intuitive easier to distinguish from other areas

@CLiersch Yes, this seems to be reasonable. I updated the issue.

commit e0408c64 (2017-07-10): Added API funtion tiglFuselageGetCrossSectionArea (#274)

The function tiglFuselageGetCrossSectionArea is committed which
computes the area of a fuselage section.  Added helper function GetArea
which computes the area of a TopoDS_Shape.

Fixes #259
closed 2017-07-10 by rainman110