DLR-SC/tigl #264: Problem within tiglFuselageGetPointAngle()

CLiersch, 2017-06-26:
I tried to use **tiglFuselageGetPointAngle()** as a faster alternative to **tiglFuselageGetPoint()**, but it seems not to work properly, if origin (y=0,z=0) is outside the fuselage. Maybe it would be better to take the cross section center as origin for the angle?
Here is an example of a fuselage tail created with tiglFuselageGetPointAngle():

[image]

[image]

One can see that from passing z=0 some (but only some) coordinates deliverd by the fuction are rather strange...

rainman110, 2017-06-26:
@CLiersch Yes indeed. This is a very awful bug.

My opinion is, that we should rethink all fuselage related getPoint functions. They are too buggy.

My personal favourite is to directly use the fuselage parameterisation. This given very fast and accurate results.

CLiersch, 2017-06-26:
I think it would be a good option to have such a fast function at hand for applications like my simple example (just generating a TECPLOT geometry file) above. I guess, if the fuselage is more or less circumferential, the resulting points might be more or less equidistant. But I could also imagine someone having to find a specific point on the fuselage (e.g. somewhere I am estimating a replacement circle taking points from circumferential position 0., 1/3 and 2/3.). Furthermore, when computing some intersections on your own, it might be very helpful to have such getPointAngle() function available.

rainman110, 2017-07-19:
As @CLiersch said, it would be good to take the cross section center as the origin for the computation. 

We can already compute a cross section wire for a given eta coordinate. Thus, we have all ingredients, to make this function work properly.

As some of the required functionality is part of the cpacs_3 branch, this will be released not before TiGL 3.

rainman110, 2017-07-19:
@joergbrech This could be also something that merlin can work on, if you have too much to do.

joergbrech, 2017-08-10:
Taking the cross section center of the segment fixes the problem. I plotted some lines of contant angle to test this.

Before (using the yz-plane origin):

[image]

After (using the cross section center):

[image]

commit 347339d9 (2017-08-15): change intersection calculation in GetPointAngle

this commit changes the intersection calculation in CPACSFuselageSegment::GetPointAngle to account for sections that are not planar.

fixes #264
commit 2f2092bf (2017-08-16): Bugfix of tiglFuselageGetPointAngle and better performance (#295)

The function GetPointAngle now takes the cross section center as starting point for the angle,
rather than the origin of the yz-plane at the current eta. 
Also, the performance of the function was improved by roughly a factor of 30. This commit
reintegrates the function getCenterOfMass from the cpacs_3 branch which avoids a planar
section, but uses a iso-parametric eta curve. The run time per call is reduced from about
39 ms to 1.2 ms.

Fixes #264
closed 2017-08-16 by rainman110