DLR-SC/tigl #275: CPACS file does not work with TIGL that works with Descartes

joergbrech, 2017-07-07:
The CPACS file is missing etaEnd in its ribs definition and cannot be opened by TIGL

**ERR 07/05 10:31:29 tigl.cpp:239] Error: Missing etaEnd or elementEndUID in CCPACSWingRibsPositioning::ReadCPACS!

ERR 07/05 10:31:29 CCPACSWingRibsPositioning.cpp:166] Missing etaEnd or elementEndUID**

[Optimale17_1.0.zip](https://github.com/DLR-SC/tigl/files/1131283/Optimale17_1.0.zip)

rainman110, 2017-07-08:
@rmaierl @codingpoets
The attached file does not contain any wing section positionings. I am not sure if this is valid?

TiGL assumes, that each section has a positioning. 

@codingpoets From the cpacs documentation, I am not sure if each section requires to be positioned or not. Can you please give me your insights?

rainman110, 2017-07-08:
The good thing is: The file works in TiGL 3 :)

So we just have to fix this for the TiGL 2 (currently master) branch.

rainman110, 2017-07-08:
The missing positionings were not the main problem.

The actual problem is line 6467 of the cpacs file:
```xml
<endElementUID>OptimaleBL3_VTP_Sec2_El1</endElementUID>
```

This is invalid CPACS! The tag's correct name is ```elementEndUID```. Hence, the line must be replaced by

```xml
<elementEndUID>OptimaleBL3_VTP_Sec2_El1</elementEndUID>
```

From my point of view, the TiGL error is correct as neither "elementEndUID" not "etaEnd" can be found and therefore the CPACS file is invalid.

TiGL 3 will provide better error messages w.r.t. CPACS loading, but it will be more foregiving in case of errors.

As this is not a TiGL issue, I'll close the issue now.

closed 2017-07-08 by rainman110