DLR-SC/tigl #286: Binaries v2.2.0 in downloads area do not include the Jar file for Java bindings

agodemar, 2017-07-31:
The binaries of Tigl v2.2.0 available here [https://github.com/DLR-SC/tigl/releases](https://github.com/DLR-SC/tigl/releases) (I'm using win64) do not include the Jar file necessary to use the Java bindings.
It seems to me a bit of an overhead having to recompile all from sources (and getting the third part libraries in place) in order to generate the Jar and test the JavaDemo.java.
Cheers

rainman110, 2017-08-01:
@agodemar This is indeed a bug and was not intentional. Self compiling the binaries should not be required.

We'll release 2.2.1 very soon as a bug fix, which will again include the java bindings,

rainman110, 2017-08-01:
This should be addresses for:
 - [ ] The Jenkins release jobs
 - [ ] The Appveyor builds

PR #287 (2017-08-01): Added java builds to appveyor
Closes #286
commit a14581bd (2017-08-01): Added java builds to appveyor (#287)

Closes #286
closed 2017-08-01 by rainman110