DLR-SC/tigl #288: Cannot compile it on windows - I get the following error

nrao1993, 2017-08-01:
I get the following error while trying to compile tigl from source on Windows 10, 64Bit.
--------------------------------------------------------------------------
-- Coverage disabled
CMake Error at TIGLViewer/CMakeLists.txt:293 (install):
  install FILES given directory "/bin" to install.
--------------------------------------------------------------------------
I've installed all the supporting libraries(opencascade and QT) and added the "\bin"s to the path as well.

rainman110, 2017-08-01:
It seems, that the shader files of OpenCASCADE are not correctly found or set by yourself.

Could you please paste the complete cmake configuration (the arguments for the cmake call) and the complete cmake output when running it from a clean build directory?

To build TiGL from scratch by yourself, I would recommend you to follow all the steps from the [appveyor.yml](https://github.com/DLR-SC/tigl/blob/master/appveyor.yml) file. This download first opencascade and configures everything like we do.

nrao1993, 2017-08-02:
I followed the steps as specified on the wiki from scratch but I somehow get the following errors.

[tiglerror.txt](https://github.com/DLR-SC/tigl/files/1193804/tiglerror.txt)

I presume this is the cmake system file

[CMakeSystem.txt](https://github.com/DLR-SC/tigl/files/1193813/CMakeSystem.txt)

This is the log

[CMakeOutputlog.txt](https://github.com/DLR-SC/tigl/files/1193816/CMakeOutputlog.txt)

The Cache

[CMakeCache.txt](https://github.com/DLR-SC/tigl/files/1193820/CMakeCache.txt)

rainman110, 2017-08-06:
Apparently, you are generating a project for VC 2010 but opening it with a later visual studio:
https://itayhauptman.wordpress.com/2016/04/10/msbuild-error-msb4019-the-imported-project-cmicrosoft-cpp-default-props-was-not-found-confirm-that-the-path-in-the-declaration-is-correct-and-that-the-file-exists-on-disk/

Apart from that, your casroot and Tixi_path  variables are wrong. Don't use the bin directory but the root directory, containing bin, lib, include ...

rainman110, 2017-08-22:
@nrao1993 Did you solve your problems? Else, I'll close the issue.

nrao1993, 2017-08-22:
No, I'm still working on it.

rainman110, 2017-08-22:
@nrao1993 I think you could ask Jonas Jepsen for help. I built TiGL by himself and is at your institure, right?

The building steps are the same for all cmake based software. Your issues are actually not TiGL specific.

nrao1993, 2017-08-22:
Yes, Jonas Jepsen is from our institute but he built TiGL on Linux and hence couldn't help me with compiling it on Windows.

nrao1993, 2017-08-22:
I get a tixi error now. I've changed the path to the binaries path as required.
```
C:\Users\Nithin Rao\Documents\DLR\tiglmakes\tigl_4\build>cmake -G "Visual Studio 10 2010" -DTIXI_PATH=C:\Users\Nithin Rao\Documents\DLR\tigl_remake\supp_tools\tixi -DCASROOT=C:\Users\Nithin Rao\Documents\DLR\tigl_remake\supp_tools\oce -DCMAKE_INSTALL_PREFIX=install ..
-- Setting build type to 'Release' as none was specified.
-- The C compiler identification is MSVC 16.0.30319.1
-- The CXX compiler identification is MSVC 16.0.30319.1
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 10.0/VC/bin/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 10.0/VC/bin/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio 10.0/VC/bin/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio 10.0/VC/bin/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:109 (find_package):
  Could not find a configuration file for package "TIXI" that is compatible
  with requested version "2.2.1".

  The following configuration files were considered but not accepted:

    C:/Program Files/TIXI 2.2.1/lib/tixi/tixi-config.cmake, version: 2.2.1 (64bit)

-- Configuring incomplete, errors occurred!
```

[CMakeOutput.txt](https://github.com/DLR-SC/tigl/files/1242370/CMakeOutput.txt)

rainman110, 2017-08-22:
You are building a 32 bit executable and try to use TiXI 64 bit. This is not going to work.

Download and use TiXI 32 bit. Also, download TiXI 2.2.4 as it is the latest release of TiXI 2.

nrao1993, 2017-08-22:
Thank you. I've successfully generated the build files.  I'm working on building it and installing it now.

closed 2017-08-24 by rainman110