DLR-SC/tigl #292: Generated manipulation API for CPACS classes

bernhardmgruber, 2017-08-08:
Currently, manipulating the object graph created by TiGL is more cumbersome than it could be. Especially creating new elements requires some repetitive arguments due to parent pointer managing and automatic uid registration.

Example (vector element): adding a new wing section might look like this:
```
tigl::CCPCASWing& wing = ...;
tigl::CTiglUIDManager& mgr = ...; // e.g. by wing.GetUIDManager()
wing.GetSections().GetSections().push_back(tigl::make_unique<tigl::CCPACSWingSection>(&mgr));
tigl::WingSection& newSection = *wing.GetSections().GetSections().back();
```
Example (optional): adding the optional structure to a component segment:
```
tigl::CCPACSWingComponentSegment& cs = ...;
tigl::CTiglUIDManager& mgr = ...; // e.g. by cs.GetUIDManager();
cs.GetStructure() = boost::in_place(&cs, &mgr);
tigl::CCPACSWingCSStructure& newStructure = *cs.GetStructure();
```
Deleting is than done by removing an element from a vector or setting an optional to boost::none.

As CPACSGen knows the constructor arguments for each type, it can generate utility methods to each class for creating instances of its children.
For vector elements we propose a method called ```AddChildName() -> Child&```, for optionals a method ```CreateChildName() -> Child&```
Similarly, ```RemoveChild(Child& child)``` for vector and ```RemoveChild()``` for optional elements can be generated.

Example (vector element): adding a new wing section might look like this:
```
tigl::CCPCASWing& wing = ...;
tigl::WingSection& newSection = wing.GetSections().AddSection());
```
Example (optional): adding the optional structure to a component segment:
```
tigl::CCPACSWingComponentSegment& cs = ...;
tigl::CCPACSWingCSStructure& newStructure = cs.CreateStructure();
```

When CreateChild() for optionals is implemented as only creating the member if it has not been created yet, otherwise just returning the reference to the existing element, we can do fun things:
```
tigl::CCPACSAircraftModel& model = ...;
model.CreateWings().AddWing().CreateComponentSegments().AddComponentSegment().CreateStructure().GetUpperShell().CreateStringer().SetStringerStructureUID("yolo");
model.CreateWings().AddWing().RemoveComponentSegments(); // removes everything unter componentSegments

auto& w1 = model.CreateWings().AddWing();
auto& w2 = model.CreateWings().AddWing(); // note that CreateWings() does nothing after the first call
auto& w3 = model.CreateWings().AddWing();
model.CreateWings().RemoveWing(w2); // reference w2 is invalid after this call
```
We at RISC have used and tested this API internally over the last month (c.f. https://github.com/RlanderRISCSW/cpacs_tigl_gen/tree/treemanip) and think it is ready to go public. The only thing we are a bit unsure of is the naming of CreateChild() for optionals as it is actually GetOrCreateIfNotThere(); however, we did not come up with a better name yet.

What do you think? Any comments, changes? I can send a PR anytime ;)

rainman110, 2017-08-10:
@bernhardmgruber Just some short feedback, such that you know that this issue is not ignored:

I am somehow ambivalent with this suggestion. I like some parts, but with other parts I have my difficulties (e.g. create..., when the Item is already there).

Please let me think about this a bit longer.

bernhardmgruber, 2017-08-10:
@rainman110 thank you for the feedback, take your time to consider

We also had a discussion about the CreateXXX() for optionals. @RlanderRISCSW argued for throwing when the element is already there. However, this would prevent chaining multiple create calls.

I look forward to your thoughts ;)

bernhardmgruber, 2017-09-11:
@rainman110 how are we doing? ;) any thoughts yet?

rainman110, 2017-09-11:
Sorry, totally forgot this "issue". I'll discuss it with a colleague and will write you on Wednesday.

joergbrech, 2017-09-14:
@bernhardmgruber We just discussed your proposal. We generally think it is a good idea to have an API, that helps in the process of creation and destruction of cpacs nodes.

We propose a small change to your first draft of the API. Your example would then be:

```cpp
tigl::CCPACSAircraftModel& model = ...;
model.GetWings(CreateIfNotExists)->NewWing() \
    .GetComponentSegments(CreateIfNotExists)->NewComponentSegment() \
    .GetStructure(CreateIfNotExists)->GetUpperShell() \
    .GetStringer(CreateIfNotExists)->SetStringerStructureUID("yolo");
model.GetWings(CreateIfNotExists)->NewWing().DeleteComponentSegments(); // removes everything unter componentSegments

auto& w1 = model.GetWings(CreateIfNotExists)->AddWing();
auto& w2 = model.GetWings(CreateIfNotExists)->AddWing(); // note that CreateWings() does nothing after the first call
auto& w3 = model.GetWings(CreateIfNotExists)->AddWing();
model.GetWings(CreateIfNotExists)->DeleteWing(w2); // reference w2 is invalid after this call
```

The benefit would be, that we don't blow up the existing classes but just add an optional argument to the getters, where you can define whether you want a new object or not.

What do you think?

bernhardmgruber, 2017-09-18:
@joergbrech I just had a talk with my colleagues @mfhava and @RlanderRISCSW and they are also of the opinion that your API design is better than my initially proposed one. I will implement it and send a PR :)

PR #313 (2017-10-24): Manipulation API for CPACS classes
fixes #292
closed 2017-10-24 by rainman110