DLR-SC/tigl #306: TiGL does not compile with OCC 7.2.0

bernhardmgruber, 2017-09-06:
When compiling TiGL with the 7.2.0 release of OpenCascade a few compile errors occur:
* GEOMAlgo_Splitter::ErrorStatus() is no longer available, i guess we should use HasErrors()
* GEOMAlgo_Splitter::myErrorStatus is no longer available, i guess we should GEOMAlgo_Splitter::myReport (myErrorStatus = 0 becomes myReport.clear())
* BRepAlgoAPI_Cut::ErrorStatus() is no longer available, i guess we should use HasErrors()

Furthermore, a few further errors occur now in TiGLViewer. Some typedefs (V3d_Coordinate, Quantity_Factor) have been replaced by Standard_Real and many methods on AIS_InteractiveContext no longer default the last boolean so it has to be specified manually.

rainman110, 2017-09-06:
OCCT 7.2.0 has currently no priority for us. We still have to make the transition to 6.9.0 (the code compiles fine, but there are some minor issues with meshes and Boolean ops ).

But feel free to make a  PR, as long as it still compiles with 6.8.0.

bernhardmgruber, 2017-09-06:
Thank you for your answer. We usually have troubles as well when upgrading OCC, but things tend to get better with each version. OCC 7.00 e.g. had a strong performance boost. Unfortunately, backward compatibility is only of little interest to us, so i might send a PR if it is part of a larger portion we contribute back, but do not count on it :/

bernhardmgruber, 2017-09-06:
btw: it seems OCC 7.2.0 now defines UNICODE and _UNICODE when configured with cmake. This causes IsPathRelative() in tiglcommonfunctions.cpp to fail to compile on windows. The Win32 API PathIsRelative() now expects a wide character string, which std::string::c_str() is not.

PR #329 (2018-01-04): Modifications to allow compiling with OpenCascade 7.2.0
Apparently fixes #306
closed 2018-01-04 by rainman110