DLR-SC/tigl #320: TIGLViewer fails auto update on Linux

codingpoets, 2017-11-30:
When an opened CPACS file is updated, the TIGLViewer usually automatically updates the displayed geometry. On Linux Mint 18.1 it only updates the view once. Every other change and saving of the opened CPACS file is not updated in the view.

The steps to reconstruct it are:
 
1. Start TIGLViewer and open a CPACS file with a geometry inside
2. Change some of the geometry parameter in the opened file and save the file
3. See how the view in the TIGLViewer is updated.
4. Perform another change in the opened CPACS file and save it
5. See how this time the geometry in the view is not updated
6. Any further modification and saving of the file is not reflected in the opened view

I have only experienced this on Linux machines. On Windows the update works without problems.

rainman110, 2017-12-01:
This is a known issue and seems to be a Qt (4?) issue on Linux only. We cannot fix this unless we are completely implementing our own way? 

Maybe it is sufficient, to use Qt5 on Linux?

@joergbrech Could you please check, wheather this issue exists also on Qt5 on Linux?

joergbrech, 2017-12-04:
I just built TIGL 2.2.1 with Qt5 on linux. It did not fix the problem.

codingpoets, 2017-12-04:
Thank you for testing. Then we will have to live with it for now. But it is not a big problem; currently just a discomfort.