DLR-SC/tigl #322: Allow flap deflection from API

joergbrech, 2017-12-01:
Currently, flap deflection is only possible from TiGL Viewer. It would be good to allow deflecting the flaps directly using the API

joergbrech, 2017-12-01:
I propose the following API functions

**Functions for wing calculations:**

```c++
TIGL_COMMON_EXPORT TiglReturnCode tiglGetControlSurfaceCount(TiglCPACSConfigurationHandle cpacsHandle,
                                                             const char * componentSegmentUID,
                                                             int * numControlSurfaces);

TIGL_COMMON_EXPORT TiglReturnCode tiglGetControlSurfaceUID(TiglCPACSConfigurationHandle cpacsHandle,
                                                          const char * componentSegmentUID,
                                                          int controlSurfaceIndex,
                                                          char ** controlSurfaceUID);

TIGL_COMMON_EXPORT TiglReturnCode tiglGetControlSurfaceType(TiglCPACSConfigurationHandle cpacsHandle,
                                                            const char * controlSurfaceUID,
                                                            TiglControlSurfaceType * type);

TIGL_COMMON_EXPORT TiglReturnCode tiglControlSurfaceGetMinimumDeflection(TiglCPACSConfigurationHandle cpacsHandle,
                                                                         const char * controlSurfaceUID,
                                                                         double * minDeflection);

TIGL_COMMON_EXPORT TiglReturnCode tiglControlSurfaceGetMaximumDeflection(TiglCPACSConfigurationHandle cpacsHandle,
                                                                         const char * controlSurfaceUID,
                                                                         double * maxDeflection);
```

I am not sure how to realize the control surface manipulation. Something like the following would be intuitive:

```c++
TIGL_COMMON_EXPORT TiglReturnCode tiglControlSurfaceGetDeflection(TiglCPACSConfigurationHandle cpacsHandle,
                                                                  const char * controlSurfaceUID,
                                                                  double * deflection);

TIGL_COMMON_EXPORT TiglReturnCode tiglControlSurfaceSetDeflection(TiglCPACSConfigurationHandle cpacsHandle,
                                                                  const char * controlSurfaceUID,
                                                                  double deflection);
```
We could store the current deflection value in  `CCPACSControlDevice` and  change ´CCPACSWing::GroupedFlapsAndWingShapes´ to get the current deflection value from each of its control devices. What do you think @rainman110?

rainman110, 2017-12-01:
@joergbrech Your proposal sounds good! Do whatever you think has to be done.

joergbrech, 2017-12-22:
Currently, `SetDeflection` only sets the value of the `CCPACSControlSurfaceDevice`, without actually changing the geometry. 

This makes sense if the function is called from the TiGL Viewer, where the transform should not be calculated each time a slider value changes. Once the flap dialog closes, `wing.GroupedFlapsAndWingShapes()` is called to apply the transform based on the deflection values.

This means we have to think about when and where we want to apply the transform when changing deflection values from the API. I see two options:

1. Add an extra argument `bool updateWingLoft` to `SetDeflection` that specifies, wether the loft of the wing should be recalculated or not. For calls from the API it should, for calls from the flaps dialog it shouldn't (or at least not before the dialog closes)
2. Call `GroupedFlapsAndWingShapes()` at a reasonable place outside of `SetDeflection`, e.g. immediately before we export to a CAD or VTK format.

What do you think @rainman110? Any preferences?

joergbrech, 2018-03-02:
Commit d0447cfcbe4d234a170fcf95675ff52b9cd5cfc1 implements the first option from my last comment and commit 6e8ef28836d0be78cce212317b8e9cea8fe036fb the second option, which should be slightly better in terms of performance