DLR-SC/tigl #371: Support for custom IGES level in CTiGLExportIges

bernhardmgruber, 2018-03-06:
Although the CTiglCADExporter is a useful abstraction to export geometry to files independent of the exported format, it is in some cases desirable to work directly with one of the derived exporters to leverage file specific features.
Such a feature is to specify custom IGES levels, when exporting using CTiglExportIges directly. We would like to have a new overload to the existing CTiglCADExporter::AddShape() which allows to specify the IGES level of the added shape:
```
class CTiglExportIges : public CTiglCADExporter {
    void AddShape(PNamedShape shape, ExportOptions options = {}); // either used from base, or overridden
    void AddShape(PNamedShape shape, int level, ExportOptions options = {}); // only available in CTiglExportIges
};
```
Can you please add such an overload? Thank you!

rainman110, 2018-03-06:
@bernhardmgruber If I understood you correctly, you want to actually attach the same iges level for all fuselage faces, another level for all wing faces etc.

Actually, this is already possible. Just set the group mode of the IGES exporter to NAMED_COMPOUNDS. This will combine all faces, which belong together into one shape and will assign each group a different iges layer.

bernhardmgruber, 2018-03-07:
Not exactly, i have several shapes which i want to group to the same IGES level.

E.g.
```
CTiglCADExporter& e = ...;
e.AddShape(s1, 1);
e.AddShape(s2, 1);
e.AddShape(s3, 7);
e.AddShape(s4, 1);
```
Shapes s1, s2 and s4 belong to level 1, s3 belongs to level 7. How can I achieve this? Thank you!

rainman110, 2018-04-05:
You can now  write custom levels as done here:
https://github.com/DLR-SC/tigl/blob/07fe27c388ae8ac5cb55408b4330fc27da46ba7b/tests/unittests/tiglExports.cpp#L283

commit 07fe27c3 (2018-04-05): Added test for custom iges levels

Closes #371
closed 2018-04-05 by rainman110