DLR-SC/tigl #376: extend python api to allow access to structure geometry

juri117, 2018-03-14:
If I get it right, in the moment its only possible to access the outer surfaces of the geometry through python api. For the cleanSky  project however we need the structure to run further analysis. Is there a way to get the TopoDS_Shape of structural parts, like ribs, stringer etc. as they are displayed in the tigl-viewer gui. Final goal will be to get an igs-file of the wing including structure.

rainman110, 2018-03-14:
The structure can now be accessed by the python interface as demonstrated in the demo:
https://github.com/DLR-SC/tigl/blob/376_wing_structure_python/examples/python_internal/example_visualization.py

We still have to provide the binaries via conda including this change.

closed 2018-03-14 by rainman110

juri117, 2018-04-21:
thank you, good to know its possible!
sry for the late reply, but is there a chance I can get this running with a cpacs 2.3 file (or upgrade my file to 3.x)?

rainman110, 2018-04-22:
@juri117 We are working on a cpacs converter that converts cpacs 2 to cpacs 3 files. It is still very much work in progress. See here:
https://github.com/DLR-SC/cpacs2to3/

PR #377 (2018-03-14): Integrated wing structure into internal python api
Integrated wing structure into internal python api and adapted visualization example to show wing structure.

The commit should not affect any other code branches than the python api.

Closes #376
commit cfeded8c (2018-03-14): Integrated wing structure into internal python api

Also, adapted visualization example to show wing structure

Closes #376