DLR-SC/tigl #403: Segment shapes do not coincide with the wing (resp. fuselage)

joergbrech, 2018-04-16:
This is only an issue as soon as the branch for #392 gets merged. Then, the wing and the fuselage get lofted with globally interpolated guide curves, while the segments get lofted with locally interpolated guide curves. It would be better if the loft of the segments where a subgeometry of the wing (resp. fuselage).

bernhardmgruber, 2018-06-29:
Are there any estimates about when this will be implemented? We just started to support guide curves in our application and this is currently our main issue^^

rainman110, 2018-07-09:
Just a note on the possible implementation:
The sections that are interpolated cut the fuselage surface at iso lines. Thus, you just have to trim the fuselage surface at the correct iso parameters to get the segment surface.

joergbrech, 2018-07-20:
**To Do** 
 - [ ] move static function `curvesToSurface` from `CTiglBSplineAlgorithms `to separate class `CTiglCurvesToSurface`
 - [ ] Implement `CTiglMakeLoft::makeLoftWithoutGuides()` using `CTiglCurvesToSurface` rather than `BRepOffsetAPI_ThruSections`
 - [ ] Abstract `makeLoft...` functions of `CTiglMakeLoft,` so that the b-splines parameters at the profiles (and guides) are provided, regardless of the algorithm ("_without guides_" vs." _Coons Patches_" vs. "_Gordon surfaces_").
 - [ ] cut the lofts at the profile parameters to obtain the segment lofts of a wing resp. fuselage