DLR-SC/tigl #407: Airfoil Y coordinates

MaloDrougard, 2018-04-17:
The y coordinates of airfoil seems to be ignored. For example the two following airfoil definitions have the same result in tiglviewer-3.
   	
	
	<wingAirfoil uID="mdAirfoil2">
          <name>mdAirfoil2</name>
          <pointList>
            <x mapType="vector">1;0;-1;0;1</x>
            <y mapType="vector">0;0;0;0;0</y>
            <z mapType="vector">0;-1;0;1;0</z>
          </pointList>
        </wingAirfoil>
	
	
	<wingAirfoil uID="mdAirfoilShifted">
          <name>mdAirfoilShifted</name>
          <pointList>
            <x mapType="vector">1;0;-1;0;1</x>
            <y mapType="vector">-10;-10;-10;-10;-10</y>
            <z mapType="vector">0;-1;0;1;0</z>
          </pointList>
        </wingAirfoil>
	
	
It's just to make sure that currently tiglviewer only support standard airfoil.

rainman110, 2018-04-17:
@MaloDrougard  Does this also happen in TiGL 2 or is this a new bug in TiGL 3?

MaloDrougard, 2018-04-17:
It happens also in tiglviewer-2.

rainman110, 2018-04-17:
@MaloDrougard We decided a while ago, that we only want to support 2d profiles. Therefore, we ignore the y - coordinate. If you want to shift the profiles, just use the positionings.

MaloDrougard, 2018-04-17:
Ok, thank you for your quick answer! 
It could be useful to have a warning when a such profile appears.

rainman110, 2018-04-17:
@MaloDrougard Good idea! A warning totally makes sense.