DLR-SC/tigl #427: Setup Travis CI for build AppImage package (portable binary for Linux)

Symbian9, 2018-05-18:
As you already provide binary for macOS & Windows and already test building process using Travis -CI, so it would be good to setup `.travis.yml` for creating AppImage package (`*.AppImage`), that could run on near all current Linux distributions without install

- http://github.com/AppImage/AppImageKit
  - https://github.com/probonopd/linuxdeployqt
  - https://github.com/probonopd/uploadtool

Setup `.travis.yml` script in the root of `master` branch, as in such examples:

- https://github.com/probonopd/linuxdeployqt/blob/master/.travis.yml
- https://github.com/leozide/leocad/blob/master/.travis.yml
- https://github.com/qpdf/qpdf/blob/master/.travis.yml
- https://github.com/jimevins/glabels-qt/blob/master/.travis.yml

Ask @probonopd if you need help in completing this task.

P.S.: Known apps that already provide AppImage's listed on AppImageHub

- https://appimage.github.io/apps

probonopd, 2018-05-18:
Apparently this is built on OBS already, and OBS has functionality to build AppImages.
https://software.opensuse.org/download.html?project=science:dlr&package=tigl

But wait -- is this a library? Then we need to make an AppImage of an app that uses this library.

rainman110, 2018-05-19:
@probonopd Thanks for your good hint to try out appimage. Indeed, creating portable linux binary for a large range of distributions is hard.

If AppImage has something to offer that solves these problems, we would be very happy.

> But wait -- is this a library? Then we need to make an AppImage of an app that uses this library.

Yes, the main part is a library, But we also have a Qt-based Viewer that uses this library. What is the best way to build and package both (using AppImage) for a large set of distros?

probonopd, 2018-05-19:
In this case, build an AppImage for the Qt-based Viewer and bundle the library into it.

rainman110, 2019-04-18:
Not planned anymore. Other priorities unfortunately

closed 2019-04-18 by rainman110