DLR-SC/tigl #431: tiglWingGetReferenceArea on TIGL_X_Z_PLANE seems not to work correctly

MaloDrougard, 2018-05-28:
When, I run the following function on the cpacs file given in the zip, I get a return value of 0. But, it seems to me that the return value should be 1. 

`tiglWingGetReferenceArea(*tiglHandle,1, TIGL_X_Z_PLANE,&bAreaXZ);`

[wing-simple-rotations-2.xml.zip](https://github.com/DLR-SC/tigl/files/2045293/wing-simple-rotations-2.xml.zip)

joergbrech, 2018-05-28:
I took a look at your example. Taking the quadrilateral points of your wing and projecting them onto the XZ-plane produces a non-convex self-intersecting quadrilateral and the formula that we use produces an area of zero. As I see it, this is not error per se. 

Consider for instance the case when one wants to check for orthogonality to the XZ-plane of a wing with a planar quadrangle. In principle, the `tiglWingGetReferenceArea `function could be used for this. In the case of orthogonality we would expect a degenerated quadrilateral in the XZ-plane with an area of zero.

We could cosider checking for degeneracy and producing a warning, but this would add some overhead to the function. @rainman110, @MaloDrougard, if there are no objections from you, I would close this issue.

MaloDrougard, 2018-05-31:
Thank you for your answer. 
Yes it make sense. 
So, in case of a degenerated quadrilateral, the area is not the absolute value of the two parts.
Can you tell me witch formula is used?

joergbrech, 2018-05-31:
We use the formula 0.5*| AC x BD |, where A,B,C,D are the four points of the quadrilateral. Unfortunately it does not return the absolute value of the two parts in the case of a self-intersecting quadrilateral.

Is this area of importance for your application?

joergbrech, 2018-05-31:
Do you know another, more general formula for the quadrilateral area, that would also work for your case?

MaloDrougard, 2018-05-31:
No, I don't know a other formula. 
I develop a application to modify CPACS files, base on TiglViewer. I use the area functions to verify that some properties are conserved after modifications. This behavior is not a issue for me. I just want to make sure that I understand it correctly to use it properly.

joergbrech, 2018-05-31:
Ok, thanks for letting us know anyway!

closed 2018-05-31 by joergbrech