DLR-SC/tigl #433: Computation of "riffled bulkhead" crashes

rainman110, 2018-05-29:
The computation of the bulkhead "riffled_bulkhead" crashes. This needs to be fixed!

Use the following cpacs file:
https://github.com/DLR-SC/tigl/blob/cpacs_3/tests/unittests/TestData/fuselage_structure-v3.xml

The issue has two aspects:
 - No geometry error should crash TiGL Viewer, even if the computation fails.
 - The compution should obviously not fail though.

bernhardmgruber, 2018-05-29:
This works fine on my PC, but I am using OCC 7.2. Maybe this is an issue with older versions?

[image]

joergbrech, 2022-02-11:
Update: It currently works with tigl  3.1.0, which still uses occt 0.17.3, but it produces the warning 

```
WRN 02/11 11:29:53 CCPACSPressureBulkheadAssemblyPosition.cpp:145] Failed to create shape for bulkhead "riffled_bulkhead", retrying planar one
```

It does *not* work in tigl3 > 3.1.0 which is based on occt 7.4, see also #878. The lofting is extremely slow.