DLR-SC/tigl #448: Shape generation of wing cells

hardt-fa-flb, 2018-07-18:
Wing cells are used to define several materials on a wing shell. In our application, we need the shapes of the wing cells to set the material distribution of the wing shell.

The current TIGL code only gives the four corner coordinates of square cells. In wing configurations like the XRF1-wing and wind turbine rotors, the spars do not have to be straight along the wing/rotor axis (figure 1), so the cells are not squares, like cell 1 in figure 2.

FA-FLB requests a method to get the OpenCASCADE shape of each cell of the wing. 

[image]

[image]

hardt-fa-flb, 2018-07-18:
The Python-code to create squared cells:

`    def _get_cell_shape(self, cell, shell_shape):
        """
        Returns the shape of a wing cell in the PreDoCS coordinate system. The cell definition works only with border input types "Eta" and "Xsi".
        
        Parameters
        ----------
        cell: tigl3.configuration.CCPACSWingCell
            The wing cell.
        shell_shape: OCC.TopoDS.TopoDS_Shape
            The shape of the wing shell that contains the cell.
        
        Returns
        -------
        OCC.TopoDS.TopoDS_Shape
            The shape of the wing cell.
        """
        inner_border = cell.get_positioning_inner_border()
        outer_border = cell.get_positioning_outer_border()
        leading_edge_border = cell.get_positioning_leading_edge()
        trailing_edge_border = cell.get_positioning_trailing_edge()
        
        if (not inner_border.get_input_type() == 0) or (not outer_border.get_input_type() == 0) or\
            (not leading_edge_border.get_input_type() == 0) or(not trailing_edge_border.get_input_type() == 0):
            raise RuntimeError('Cell definition only with border input types "Eta" and "Xsi".')
        
        eta1 = inner_border.get_eta_2_choice1()
        xsi1 = trailing_edge_border.get_xsi_1_choice2()
        eta2 = outer_border.get_eta_2_choice1()
        xsi2 = trailing_edge_border.get_xsi_2_choice2()
        eta3 = outer_border.get_eta_1_choice1()
        xsi3 = leading_edge_border.get_xsi_2_choice2()
        eta4 = inner_border.get_eta_1_choice1()
        xsi4 = leading_edge_border.get_xsi_1_choice2()
        
        bb = Bnd_Box()
        brepbndlib.Add(shell_shape, bb)
        Xmin, Ymin, Zmin, Xmax, Ymax, Zmax = bb.Get()
        
        pnt1 = self._component_segment.get_point(eta1, xsi1)
        pnt2 = self._component_segment.get_point(eta2, xsi2)
        pnt3 = self._component_segment.get_point(eta3, xsi3)
        pnt4 = self._component_segment.get_point(eta4, xsi4)
        pnt1.SetZ(Zmin)
        pnt2.SetZ(Zmin)
        pnt3.SetZ(Zmin)
        pnt4.SetZ(Zmin)
    
        wire = BRepBuilderAPI_MakeWire(BRepBuilderAPI_MakeEdge(pnt1, pnt2).Edge(),
                                    BRepBuilderAPI_MakeEdge(pnt2, pnt3).Edge(),
                                    BRepBuilderAPI_MakeEdge(pnt3, pnt4).Edge(),
                                    BRepBuilderAPI_MakeEdge(pnt4, pnt1).Edge()).Wire()
        face = BRepBuilderAPI_MakeFace(wire).Face()
        
        #prism_direc = gp_Dir(0,0,1)
        prism_direc = gp_Vec(0,0,Zmax-Zmin)
        cell_prism = BRepPrimAPI_MakePrism(face, prism_direc).Shape()
        return BRepAlgoAPI_Common(cell_prism, shell_shape).Shape()`

rainman110, 2018-07-18:
@hardt-fa-flb I just called @rmaierl  from Airbus. They already implemented the cell geometry and he will send us the code for it. Then we just have to integrate it back into TiGL.

ewerthen, 2018-07-18:
Perfect! What do you think how long does is take? Thank you

rainman110, 2018-07-18:
I can't answer this right now, since I haven't seen the code yet. I guess, I won't take very long.

rainman110, 2018-07-18:
I just tried to integrate the airbus code for the wing cell shape computation. I had problems with it, as suddenly the member CPACSWingCell::m_skin seemed to become optional.

In our TiGl code, this is not the case. @bernhardmgruber Did you change the cpacs schema or is the skin now optional in a newer version?

rainman110, 2018-07-19:
The wing cell geometry code seems to be working:
[image]

This is the the wing cell defined in the [cell_rib_spar_test.xml](https://github.com/DLR-SC/tigl/blob/cpacs_3/tests/unittests/TestData/cell_rib_spar_test.xml) example.

What remains to be done is to wrap the code into the python api.

hardt-fa-flb, 2018-07-19:
Looks good! This is exactly what we need!

rainman110, 2018-07-23:
@hardt-fa-flb + @ewerthen The wing cell geometry code is now in the main branch. I will create some conda packages the next days. 

One important detail: I changed the api of the GetEtaXsi methods! They are returning now an EtaXsi structure instead of a tuple. This is a cleaner and more scalable solution. Unfortunately, you have to change you python code a bit.

closed 2018-07-23 by rainman110

rainman110, 2018-07-24:
@hardt-fa-flb I uploaded the new conda windows packages. You can try out the functionality. Just use ```conda update tigl3```

commit 52aea1c0 (2018-07-19): Made Wing Cell a geometric component

This automatically registers the wing cell at the uid manager
and is therefore available in the api!

Closes #448
commit 017946ac (2018-07-23): Made Wing Cell a geometric component

This automatically registers the wing cell at the uid manager
and is therefore available in the api!

Closes #448
commit fb32774c (2018-07-23): Return correct type of ribs, spars and cells in python layer

Closes #448
PR #450 (2018-07-23): Integrated cell geometry code from airbus
I integrated the cell geometry from airbus.

The following changes to the original code have been made:
 - Adaptations, since CPACSWingCell::m_skin is not optional in our code
 - Made the cell a geometric component
 - Integrated into the Python wrapper
 - Added Python example

@bernhardmgruber , @rmaierl  Can you please commit a description of CCPACSWingShell::SparSegmentsTest? I don't know what this function is for. Is it possible, to extract this method into a separate function?

Closes #448