DLR-SC/tigl #449: Review GLOG initialization and shutdown

rainman110, 2018-07-18:
TiGL initialzes GLOG in CTiglLogging
https://github.com/DLR-SC/tigl/blob/84836e6b24ac91d92c8e85594609e2c50005dfa6/src/logging/CTiglLogging.cpp#L98

and shuts in down in its destructor in
https://github.com/DLR-SC/tigl/blob/84836e6b24ac91d92c8e85594609e2c50005dfa6/src/logging/CTiglLogging.cpp#L63

If an application that uses TiGL itself also uses GLOG, this will cause problems. Hence, these routines must be called outside of TiGL.

How can we make sure, that the logger is correctly initialized when used by TiGL?

bernhardmgruber, 2018-07-18:
we have exactly this issue in Descartes (internal software based on TiGL). Our current workaround is to comment out the respective lines in TiGL and call them ourselves in our main function.

joergbrech, 2023-12-15:
I suppose we could make this a cmake option? `cmake .. -DTIGL_DISABLE_GLOG=ON`