DLR-SC/tigl #487: TiGL Viewer: Make default colors and materials configurable

rainman110, 2018-09-04:


Mk-arc, 2020-08-13:
Hi Martin,

can you give a little more details about this issue? What are the relevant classes / files? And what are the relevant properties?

rainman110, 2020-08-13:
The colors and material of a shape display in TiGL Viewer is set here during the displayShape call:
https://github.com/DLR-SC/tigl/blob/master/TIGLViewer/src/TIGLViewerContext.cpp#L320

To make default colors and materials configurable, some options have to be added to the TiGLViewerSettings Class and Dialog:
https://github.com/DLR-SC/tigl/blob/master/TIGLViewer/src/TIGLViewerSettings.h
https://github.com/DLR-SC/tigl/blob/master/TIGLViewer/src/TIGLViewerSettingsDialog.h

The settings class stores the (default) values, the dialog allows to modify these values.

Ideally, those values can also be modified using TiGL Viewer's scripting console. The most important class is TiGLViewerScriptEngine, which allows to access all QObject based classes via the console. It defines the global ``app`` object, that is a handle to the application windows.

What I can think of is setting the shape color / material like follows:

```js
// set color red
app.settings.setShapeColor(QColor(1.0, 0.0, 0.0));
app.settings.setShapeMaterial(Material_Enum_Value);
```

By the way, typing ``help(app)`` already shows what is working right now.

Mk-arc, 2020-08-15:
Hi,

so i started implementing this feature. I already added 1 parameter (shapeColor) to the settings, the settings dialog and exposed the settings in the scripting engine:

[image]

[image]

However, there is still one Problem. Maybe someone can help me here. The ScriptEngine seems to not know the QColor datatype. Which results in an error when trying to use it:

[image]

I guess theres some sort of "Register Meta Type" method for the scripting engine, where I have to register the QColor type?

Mk-arc, 2020-08-15:
A workaround could also be to expose additional public slot methods for the settings which take int arguments:

`void setShapeColor(int r, int g, int b, int a = 0);`
->
`app.settings.setShapeColor(255,0,255,255)`

Which works quite well and is also easier to write in the Console

rainman110, 2020-08-15:
Hi! Awesome work!

Using r,g,b values is totally fine and even more user friendly. When I implemented ``app.viewer.setBackgroundColor`` to temporarily change the background color, I also used r,g,b. 

Wrapping QColor is probably possible, but too much of a hassle.

Not the the requirements list here, but I'd also like to configure the color of the "Symmetry Shapes". When loading an aircraft, symmetric components (e.g. left wing) is displayed in a different color. 

It would be awesome, if the user can also configure this.

Mk-arc, 2020-08-16:
Looking quite fine now already. I will open the Pull Request soon.

One more question, when someone is setting the shapeColor (or symmetrie color / Material whatever) should it directly change the color of already loaded models. Or should it really just be a "default" color.. meaning only changing the color of newly loaded models.

In the second case.. I would suggest really calling the methods like:

`app.settings.setDefaultShapeColor(...) `
and not 
`app.settings.setShapeColor(...)`

Any methods sounds fine to me, what behavior do you expect?

rainman110, 2020-08-16:
It should only change the color of shapes the will be displayed after this change.

I am okay changing the name including "default".

PR #734 (2020-10-14): Make default colors and materials configurable
Make default colors and materials configurable

## Description
Added default shape material (for normal shapes and symmetry shapes) and default material to the settings and the settings Dialog. Exposed some settings to the TiglScriptEngine. And some 

## How Has This Been Tested?
Run TiglViewer, change settings via Dialog and Script. Load Plane Model, check colors and material. Change Settings via Dialog or Script. Reload Model to see if Materials changed on the model.

## Screenshots, that help to understand the changes(if applicable):
[image]
[i [... cut]
closed 2020-10-14 by rainman110