DLR-SC/tigl #501: F6 Fuselage loft fails to build since new code changes

rainman110, 2018-09-14:
[DLR-F6-150_WBHV_LC-v3_section_removed.zip](https://github.com/DLR-SC/tigl/files/2383848/DLR-F6-150_WBHV_LC-v3_section_removed.zip)

joergbrech, 2018-09-14:
There seems to be a Problem with ``CTiglCurvesToSurface``: The surface generated from the fuselage profiles seems broken, and therefore TiGL cannot make a solid out of the fuselage:
[image]

I saved the broken fuselage Shell and the Profile curves used in the Interpolation as breps (see zip file), but I haven't checked yet what exactly goes wrong
[fuselage_lofting_bug.zip](https://github.com/DLR-SC/tigl/files/2384130/fuselage_lofting_bug.zip)

rainman110, 2018-09-14:
Strange. I am sure it worked with curvestosurface before. I'll have a look into it.

commit 9a287b88 (2018-09-21): Fixed common knot vector creation

Fixes #501
PR #503 (2018-09-22): Jk 501 fuselage lofting bug
fixes #501
closed 2018-09-22 by rainman110