DLR-SC/tigl #508: Compilation with OpenCASCADE 7.0.0 or newer

agodemar, 2018-09-20:
In order to integrate my work with java and OpenCASCADE functions, which may go possibly in `bindings/java_internal`, I would like to compile TiGL with OCCT instead of OCE. 

Any indications on how to achieve this?
Thanks

rainman110, 2018-09-20:
Hi @agodemar! Sure this works!

Just configure cmake with CASROOT=<OCCT Directory>. It should find OpenCASCADE.

You can also explicitly disable to search for OCE via the ``OpenCASCADE_DONT_SEARCH_OCE=ON`` option. 

Finally, if TiGL does not find OpenCASCADE on  its own, you can still define both
```
OpenCASCADE_INCLUDE_DIR
OpenCASCADE_LINK_DIRECTORY
```

I hope it works!

agodemar, 2018-10-08:
@rainman110 Not yet, actually. I'll report some local adjustments that were necessary to get it working. I'll report later on.

agodemar, 2019-04-05:
Issue solved. Compilation works.

closed 2019-04-05 by agodemar