DLR-SC/tigl #534: Evaluate, whether conan could be used for dependency management

rainman110, 2018-11-14:
This talk sums everything up quite nicely:

https://www.youtube.com/watch?v=S4QSKLXdTtA

bernhardmgruber, 2018-11-15:
There is a multitude of package managers out there. At some point I thought about adding TiXI to vcpkg, which I use privatly. Jason Turner also has a short intro to hunter, vcpkg and conan, which might be interesting to you:

- https://www.youtube.com/watch?v=9cCQHJ-cNHY
- https://www.youtube.com/watch?v=O2_N8OzPGWQ
- https://www.youtube.com/watch?v=KOeOLOu6nHw

I do not want to recommend any of those to you. I played only with vcpkg so far and the Visual Studio integration works well, but for cmake you have to pass an annoying cmd line argument.

I am interested in what you will pick ;)

rainman110, 2018-11-15:
Here's a nice list of pros and cons of the different package managers:
https://www.reddit.com/r/cpp/comments/8t0ufu/what_is_a_good_package_manager_for_c/e141ggo/

I also like to add:
 - spack (https://spack.io/)
 - conda (https://conda.io/docs/index.html)

With conda we already have a lot of (mixed) experience but in the end it works pretty well. What conda does really well is to support relocation of binary packages by replacing the original installation directory during build with the user installation directory.

bernhardmgruber, 2018-11-21:
Vcpkg works for Linux as well. And I believe you are right, that you can only have one version (usually the latest) installed. Vcpkg installs packages system wide, which is usually what I want. But yes, if you want to test with different versions, you still have to maintain those manually.

Thank you for the pros/cons list. It is brief but excellent!

Have you thought about providing support for multiple package managers?

rainman110, 2018-11-21:
I don't really want to support package managers that I don't use, since this is a big maintenance overhead.

I experienced this with the support for several bindings. I cannot guaranty die quality of bindings, which I don't (want) to test. The same if true for packaging system that we don't use.

Right now, I already have to maintain a couple of packaging systems, which are:
 - conda for the python builds
 - Debian packages
 - RPM packages

ras0219-msft, 2018-11-22:
For vcpkg, you can have different copies of packages by simply having different vcpkg instances -- each "clone" of vcpkg is a totally separate package graph. We do have the option to "promote" one of them to be user-wide for convenience, but all the other operations are totally local. We view it as a "best of both worlds approach" -- convenient when possible, capable when not :)

Additionally, each "commit" to the vcpkg repo is like a lockfile for the universe, specifying exactly what versions of every package work together (and we track the latest stable versions upstream). You can modify what packages to use by simply replacing the corresponding entry in `ports/` with your own version.

bernhardmgruber, 2018-11-30:
@ras0219-msft thank you for the insight, I did not think about having multiple installations of vcpkg, which might be a useful concept!

An advantage/disadvantage of vcpkg is also that it builds all the dependencies from source. This is awesome because you can then debug into third party libraries. On the other hand, if for some reason the sources fail to download or compile, you are stuck. One might try to update to some older revisions, but this is tedious. I just had two issues this week with a hash missmatch on the 7zip version and a build fail on OpenSSL for x86-windows.

I have just seen a talk about conan at CppCon 2018 and it also looks good. It seems to be more invasive into your CMake scripts though and has less available packages. But it supports binary distribution, so it might be more stable.

In the end, the decision is yours to make @rainman110 ;)

rainman110, 2018-12-05:
@ras0219-msft Thanks for the information. I understand that you guys are trying to boost the vcpkg ecosystem and I think it has a very good potential.

For now, vcpkg has - from the practical point of view - some major limitations, as @bernhardmgruber  already mentioned, namely
 - Missing versioning support
 - Missing support for binary file distribution

We are using a pretty large third-party library that builds up to an hour. Thus, only source distribution increases the time-to-solution by far. Versioning is also very important for us. We are using patches sources of the third party library and experimenting with several version of it. To be restricted to only one version is too limiting for us right now.

@bernhardmgruber Conan used to be invasing the cmake scripts in the past. Now you can use it completely transparently without modifying the CMakeLists. Otherwise, I would not use it.

bernhardmgruber, 2018-12-07:
@ras0219-msft You don't happen to be this guy: https://www.youtube.com/watch?v=sBP17HQAQjk

rainman110, 2018-12-07:
Sure, that's him ;)