DLR-SC/tigl #540: Fuselage misbuild  when using relative positionings

cfsengineering, 2018-11-23:
The construction of the fuselage seems not correct if we use positioning relatively from a section that is not the previous one. 
This bug should be introduced recently because it is not present in older version. The picture shows the same aircraft  displayed in tiglviewer-2 and tiglviewer-3. The cpacs of the aircraft can be found in the attached file. 

[image]

[simple-aircraft-fuselages.xml.zip](https://github.com/DLR-SC/tigl/files/2610219/simple-aircraft-fuselages.xml.zip)

rainman110, 2018-11-23:
Thanks for the report! I'll have a look into it.

rainman110, 2018-11-23:
The error already exists in TIGL 2 as well. The difference is:
 - TiGL 2: shows fuselage segments
 - TiGL 3: shows fused fuselage

Hence, there is a bug when computing the fused fuselage. Strangely, the position of the segments is okay.

rainman110, 2018-11-23:
@cfsengineering You could have added, that you are changing the order of segments. This is right now in deed not supported.

cfsengineering, 2018-11-23:
Yes, you are right the unordered segments are the cause of the trouble. Sorry, I didn't noticed it before. I update the issue. But strangely, the older version (TiGL 3.0.0 RC1) was returning the correct values when I use "tiglFuselageGetSectionCenter" in my tests  and now is no more the case.

PR #542 (2018-11-28): Fixed unordered fuselage segments
When the segments are in wrong order,
a warning is printed and the segments will be ordered.

Fixes #540
commit dfc93d8d (2018-11-28): Fixed unordered fuselage segments

When the segments are in wrong order,
a warning is printed and the segments will be ordered.

Fixes #540
closed 2018-11-28 by rainman110