DLR-SC/tigl #549: TiGL Viewer: Brand Screenshots with the TiGL Logo

rainman110, 2018-12-10:
I often found screenshots in the web of tigl generated aircraft. We should make sure that these contain the TiGL logo such that we get the appropriate acknowledgement.

Mk-arc, 2020-08-05:
I created a first prototype, how should the logo be integrated (which file?, position, size, opacity).

Here are 2 different versions of the current implementation:

1)
[image]

2)
[image]

rainman110, 2020-08-05:
Hi @Mk-arc ! This is awesome 😄 . Great to have you back as a contributor!

I hope you are doing fine!

I personally like version 2 more, even though it's easier to cut away the logo.

If you want, just create a PR.

Mk-arc, 2020-08-05:
Thansk @rainman110 ! I'm doing fine! I hope you are also doing fine.

is there a way to run "make checkstyle" on linux?

rainman110, 2020-08-05:
``make checkstyle`` should work. You might need to initialize the git submodules first as @joergbrech wrote here:
https://github.com/DLR-SC/tigl/pull/728#issuecomment-669192299

PR #728 (2020-08-05): #549: implemented Screenshot watermark feature
closes #549

Watermark Screenshots with tigl-logo taken with TiGLViewer

- added a transparent tigl logo to ressources. 
- draw tigl logo onto screenshots when they are being created

## Description

TiglViewer Screenshots are now watermarked with the tigl-logo (bottom right corner).
main changes in TIGLViewer/src/TIGLViewerWidget.cpp

## How Has This Been Tested?
Tested locally on ubuntu 20.04. Created Screenshots with various tiglViewer scalings. 
width > 1024
width < 100

## Screenshots, that help to understand the changes(if applicable):
[imag [... cut]
commit 53444819 (2020-08-05): #549: implemented Screenshot watermark feature (#728)

Added tigl-logo to screenshots from TiGLViewer

Fixes #549
closed 2020-08-05 by rainman110