DLR-SC/tigl #552: Regard parent UId of generic geometry component

rmaierl, 2018-12-17:
The engines of the OptiMALE CPACS file have the fuselage as their parent, but the fuselage positioning is not regarded during the geometry generation.

rainman110, 2019-01-31:
This is a TiGL 2 related issue.

@rmaierl Do you need this to be fixed in TiGL 2? In TiGL 3, this is already fixed (see attached screenshots)

TiGL2:
[image]

TiGL3:
[image]

rmaierl, 2019-02-01:
If it is working in TiGl3 it is fine for me.
The user can easily adapt the positioning node of the generic geometry, if the use of TiGl 2 is required.