DLR-SC/tigl #577: For a fuselage, a GetSection(std::string uid) would be useful

joergbrech, 2019-03-13:


bernhardmgruber, 2019-03-17:
I uses to see many of these functions for getting subelements by UID. I always wondered why you need them. You could just query the UidManager which is available almost everywhere. You could save a few functions this way.

joergbrech, 2019-03-18:
My intention was to have consistent getters, since these functions are available in other cpacs lists, such as the fuselage/wing segments. But you make a good point, these functions are probably just clutter

rainman110, 2019-03-18:
The problem with the current uidmanager is, that it is right now not possible to check, whether one node is parent of another.

This would be required, when you want e.g. a section from a specific fuselage, and not from another.

bernhardmgruber, 2019-03-18:
But since you pass the UID as argument, which is globally unique, the section is uniquely identified and retrieved, whether you use the UidManager or a GetSection(uid). So the effect is the same in the "good" case.

The only advantage I can see is that you cannot wrongly retrieve a section, which does not belong to a fuselage/wing. But if this is the case (and the UID comes from another element in CPACS referring to this section, e.g. a spar), then the CPACS design is faulty. GetSection(uid) could detect this case.

commit 4c21c20a (2019-03-18): Add GetSection(const std::string& sectionUID) function for fuselage and wing sections

fixes #577
PR #578 (2019-03-22): Add GetSection(const std::string& sectionUID) function
 for fuselage and wing sections

fixes #577
closed 2019-03-22 by joergbrech