DLR-SC/tigl #582: TiglViewer: can not build fuselage

cfsengineering, 2019-03-19:
TiglViewer

rainman110, 2019-03-19:
? Can you please add some detail?

cfsengineering, 2019-03-19:
done :)

rainman110, 2019-03-19:
I can confirm this bug. I'll have a look into it tomorrow.

joergbrech, 2019-03-19:
Good find! I just ran your example in debug mode, it crashes here: 

https://github.com/DLR-SC/tigl/blob/bf998a86a8e44744504becd406f980e328742565/src/geometry/CTiglPatchShell.cpp#L169

There still seems to be an issue with the loft generation sometimes, we will take a look at it.

rainman110, 2019-03-19:
@joergbrech Since this is your code, could you have a look into this issue?

joergbrech, 2019-03-19:
Ok, so this is the situation:
 - In CTiglMakeLoft, the generated fuselage loft is cut at the profile parameters, so that the loft consists of patches exactly as it would if it were generated from a curve network. This makes sense because we want to have consistent shapes independent of the algorithm used to create them.
 - The function CutShellAtUVParamters calls CTiglBSplineAlgorithms::trimSurface, which is just a wrapper for the corresponding OpenCascade function + an OpenCascade conversion. It cuts the input face at the desired parameters. So far, the parameters passed to the function seem to be correct, but one of the trimmed surfaces (the fourth segment in this specific example) calculated by this function is way too large.
 - As a consequence, faces of the loft overlap and we cannot make it a solid.

This is more of a note to self, I will check out what goes wrong exactly when I find the time later this week.

joergbrech, 2019-03-20:
This seems to be a bug in OpenCascade to me, at least I can reproduce it in pythonocc (The input surface is still generated by TiGL though...)

### TiGL
[image]

### pythonocc
[image]

### brep file and python script
[fuselabe_bug.zip](https://github.com/DLR-SC/tigl/files/2989338/fuselabe_bug.zip)

rainman110, 2020-03-04:
I condensed this issue and posted it at the OCCT mantis:
https://tracker.dev.opencascade.org/view.php?id=31402

It even exists with OpenCASCADE 7.4.0.

rainman110, 2020-03-04:
The issue should now be fixed with my workaround

closed 2020-04-22 by rainman110

rainman110, 2020-11-13:
Opencascade has finally fixed this bug:

https://tracker.dev.opencascade.org/view.php?id=31402

As we have a workaround in place, there is nothing we have to do right now. I am writing this just for documentation purposes.

commit 118e6407 (2020-03-04): Implemented workaround for surface trimming bug

There is an issue in Geom_BSplineSurface::Segment, that leads
to a wrong trimming, if the trimming parameter is very close
to a knot.
This workaround checks, whether we are near a knot in the tolerance
of 1e-10. If yes, we set the trimming parameter to the knot.

Fixed #582
commit bea8d54d (2020-04-22): Implemented workaround for surface trimming bug

There is an issue in Geom_BSplineSurface::Segment, that leads
to a wrong trimming, if the trimming parameter is very close
to a knot.
This workaround checks, whether we are near a knot in the tolerance
of 1e-10. If yes, we set the trimming parameter to the knot.

Fixed #582
commit 35b7b854 (2020-07-14): Implemented workaround for surface trimming bug

There is an issue in Geom_BSplineSurface::Segment, that leads
to a wrong trimming, if the trimming parameter is very close
to a knot.
This workaround checks, whether we are near a knot in the tolerance
of 1e-10. If yes, we set the trimming parameter to the knot.

Fixed #582