DLR-SC/tigl #584: How to install on python

CedricDPS, 2019-03-29:
Hello i try to install tigl on my python interpreter.
After install it from conda, i debug my code and i obtain the following error :

`Traceback (most recent call last):
  File "C:\PROGRAMS\Miniconda3\lib\site-packages\tigl3\tigl3wrapper.py", line 278, in Tigl3
    lib = ctypes.cdll.tigl3
  File "C:\PROGRAMS\Miniconda3\lib\ctypes\__init__.py", line 426, in __getattr__
    dll = self._dlltype(name)
  File "C:\PROGRAMS\Miniconda3\lib\ctypes\__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 126] Le module spécifié est introuvable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\cedric.dalaudiere.DPS\.p2\pool\plugins\org.python.pydev.core_7.1.0.201902031515\pysrc\pydevd.py", line 2225, in <module>
    main()
  File "C:\Users\cedric.dalaudiere.DPS\.p2\pool\plugins\org.python.pydev.core_7.1.0.201902031515\pysrc\pydevd.py", line 2218, in main
    globals = debugger.run(setup['file'], None, None, is_module)
  File "C:\Users\cedric.dalaudiere.DPS\.p2\pool\plugins\org.python.pydev.core_7.1.0.201902031515\pysrc\pydevd.py", line 1560, in run
    return self._exec(is_module, entry_point_fn, module_name, file, globals, locals)
  File "C:\Users\cedric.dalaudiere.DPS\.p2\pool\plugins\org.python.pydev.core_7.1.0.201902031515\pysrc\pydevd.py", line 1567, in _exec
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "C:\Users\cedric.dalaudiere.DPS\.p2\pool\plugins\org.python.pydev.core_7.1.0.201902031515\pysrc\_pydev_imps\_pydev_execfile.py", line 25, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "H:\PROJET\KARREN\CEASIOM_CPACS\RCE\CEASIOMpy\test\TestCPACS2SUMO\test_cpacs2sumo.py", line 26, in <module>
    from lib.CPACS2SUMO.cpacs2sumo import convert_cpacs_to_sumo
  File "H:\PROJET\KARREN\CEASIOM_CPACS\RCE\CEASIOMpy\lib\CPACS2SUMO\cpacs2sumo.py", line 32, in <module>
    from lib.utils.cpacsfunctions import open_tixi, close_tixi
  File "H:\PROJET\KARREN\CEASIOM_CPACS\RCE\CEASIOMpy\lib\utils\cpacsfunctions.py", line 38, in <module>
    import tigl3.tigl3wrapper as tigl3wrapper
  File "C:\PROGRAMS\Miniconda3\lib\site-packages\tigl3\tigl3wrapper.py", line 269, in <module>
    class Tigl3(object):
  File "C:\PROGRAMS\Miniconda3\lib\site-packages\tigl3\tigl3wrapper.py", line 285, in Tigl3`

joergbrech, 2019-03-30:
Hi Cedric, can you please tell me the command you used to install TiGL and which python version you are using? Can you run our examples, e.g. https://github.com/DLR-SC/tigl/blob/cpacs_3/examples/python/api_basic_demo.py?

CedricDPS, 2019-04-01:
Hi Jan, thank you for your quick answer. 

I 'm using Python 3.7 (x64) from Miniconda 4.6.8 on Windows 7 Professional 64 bits.

To install TiGL, I have downloaded the win-64/tigl3-3.0.0rc1-py35_vc14_9.tar.bz2 file available on https://anaconda.org/DLR-SC/tigl3/files and I have installed it with conda command : `conda install "<path>\win-64/tigl3-3.0.0rc1-py35_vc14_9.tar.bz2"`

Concerning your test code, I have tried to run it but it seems that I have the same issues. After debugging, problems come from the import line : `from tigl3.tigl3wrapper import Tigl3, TiglBoolean`

Please find attached the python error :

`Traceback (most recent call last):
  File "C:\PROGRAMS\Miniconda3\lib\site-packages\tigl3\tigl3wrapper.py", line 278, in Tigl3
    lib = ctypes.cdll.tigl3
  File "C:\PROGRAMS\Miniconda3\lib\ctypes\__init__.py", line 426, in __getattr__
    dll = self._dlltype(name)
  File "C:\PROGRAMS\Miniconda3\lib\ctypes\__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 126] Le module spécifié est introuvable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\cedric.dalaudiere.DPS\.p2\pool\plugins\org.python.pydev.core_7.1.0.201902031515\pysrc\pydevd.py", line 2225, in <module>
    main()
  File "C:\Users\cedric.dalaudiere.DPS\.p2\pool\plugins\org.python.pydev.core_7.1.0.201902031515\pysrc\pydevd.py", line 2218, in main
    globals = debugger.run(setup['file'], None, None, is_module)
  File "C:\Users\cedric.dalaudiere.DPS\.p2\pool\plugins\org.python.pydev.core_7.1.0.201902031515\pysrc\pydevd.py", line 1560, in run
    return self._exec(is_module, entry_point_fn, module_name, file, globals, locals)
  File "C:\Users\cedric.dalaudiere.DPS\.p2\pool\plugins\org.python.pydev.core_7.1.0.201902031515\pysrc\pydevd.py", line 1567, in _exec
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "C:\Users\cedric.dalaudiere.DPS\.p2\pool\plugins\org.python.pydev.core_7.1.0.201902031515\pysrc\_pydev_imps\_pydev_execfile.py", line 25, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "H:\PROJET\KARREN\CEASIOM_CPACS\RCE\CEASIOMpy\test\python_demo.py", line 19, in <module>
    from tigl3.tigl3wrapper import Tigl3, TiglBoolean
  File "C:\PROGRAMS\Miniconda3\lib\site-packages\tigl3\tigl3wrapper.py", line 269, in <module>
    class Tigl3(object):
  File "C:\PROGRAMS\Miniconda3\lib\site-packages\tigl3\tigl3wrapper.py", line 285, in Tigl3
    "  1) The PATH (Windows) / LD_LIBRARY_PATH (Linux) environment variable points to the library\n" +
Exception: Could not load the tigl3 library. Please check if:
  1) The PATH (Windows) / LD_LIBRARY_PATH (Linux) environment variable points to the library
  2) The architecture of the library matches the architecture of python (a 32 bit python needs a 32 bit shared library)`

Thank you in advance,

rainman110, 2019-04-01:
@CedricDPS You've installed tigl3 incorrectly. Tigl3 depends on OpenCASCADE and other software. When explicitly installing a downloaded package, in won't install its dependencies. Therefore you get the message above.

At the moment, we only support Python 3.5. To install tigl correctly, enter on the anaconda prompt:

```
conda create -n tigl_env tigl3 python=3.5
```

This will create a virtual environment "tigl_env" and installs all required packages into this environment.

CedricDPS, 2019-04-01:
Hi Martin, it works.
Thank you for your time,
Cedric

closed 2019-04-01 by joergbrech