DLR-SC/tigl #607: Parent Translation not respected when drawing component segment shapes

rainman110, 2019-06-04:
This happens e.g., when we draw a wing structure. Then, the spars and ribs are at the correct position, the wing shape (component segment) is not.

This brings up the question, why the component segment has its own shape and not uses the wing shape?

sdeinert, 2019-06-05:
There could be more than one component segment per wing.

rainman110, 2019-06-05:
Sure, the component segment is part of the wing. But the wing shape should include all component segments. So again, do we really need the component segment shape are is the wing shape sufficient?

sdeinert, 2019-06-05:
Yes then you use the get wing shape method. If you want the shape of the component segment you use the other. E.g. you have different shell properties in all you component segments and want to calculate the mass of the skin. Then you would like to have the shape of only a specific component segment.