DLR-SC/tigl #610: Robustness of wing structure

joergbrech, 2019-07-10:
The generation of the wing structure relies on Boolean operations, e.g. when calculating the intersection of a plane with the wing skin to generate a spar. Currently, this generation seems to be not very robust and highly dependent on the sparPositions _(an example file has been sent by email)_.

bernhardmgruber, 2019-07-10:
I have been away from TiGL for almost a year now, so take this with a big grain of salt:
I believe you still work mostly with OCE 6.x. When I used TiGL with OCC 7.1 and later 7.3, I noticed majer improvements in reliabilty and speed. Therefore, your problem could be solved in newer versions of OCC. Since OCE is lagging behind OCC, maybe a fix is still necessary.

joergbrech, 2019-07-10:
Thanks for the input! This is going to be the first thing I will try.

rainman110, 2019-07-10:
@bernhardmgruber This is also what I suggested!

There are some tests failing though with the new OpenCASCADE 7.3. But I am confident, that we will figure out how to fix them.

The main challenge will be, to upgrade pythonOCC to OpenCASCADE 7.3. Fortunately, Thomas Paviot already started working on this.

rainman110, 2020-05-12:
relates to issue #679 , closing this for now

closed 2020-05-12 by rainman110