DLR-SC/tigl #613: Can't open CPACS configuration

Ofnir93, 2019-07-18:
Hello everybody, this is my first time writing here so I hope that this kind of topic is okay.

My problem is that my MATLAB won't open a CPACS configuration using the function '_tiglOpenCPACSConfiguration_' and will always return me this error: '
**Error: "tiglOpenCPACSConfiguration" returned code 11'**

As I am relatively new to programming I cannot understand what this error means and so I have no idea what to do.

This is the really simple MATLAB script that doesn't seem to work:

cd(pwd)
IN_file=strcat(pwd,'\Database.xml');     
tixi_h=tixiOpenDocument(IN_file);
tiglHandle = tiglOpenCPACSConfiguration(tixi_h, '');

rainman110, 2019-07-19:
Hi,

the error code 11 means "Wrong CPACS Version", see here:
https://github.com/DLR-SC/tigl/blob/cpacs_3/src/api/tigl.h#L66

You are probably trying to open a cpacs 2 file with tigl 3. This won't work, therefore the error message.

You either need TiGL 2 to open this file, or convert the CPACS 2 file to CPACS 3 using our tool cpacs2to3: https://github.com/DLR-SC/cpacs2to3

If you need more advice, don't hesitate to ask.

Ofnir93, 2019-07-20:
Thank you very much, I was able to run the script by downgrading tixi to a 2.x version. 

I would also like to thank you for linking me the explaination of the error codes so now I will be a little bit more able to solve them by myself in the future.

You have been a great help!

closed 2019-07-20 by Ofnir93