DLR-SC/tigl #614: wingGetWettedArea() error

ACdesigner, 2019-07-26:
I am currently using tigl wrapper for python/matlab, the tigl.wingGetWettedArea(wingUID) gives problem while tigl.wingGetSpan(wingUID) is fine. Does anyone have any idea why? Version tigl 2.2.3, tixi TIXI 3.0.3. Thanks in advance.

        errorCode = self.lib.tiglWingGetWettedArea(self._handle, _c_wingUID, ctypes.byref(_c_wettedAreaPtr))
        catch_error(errorCode, 'tiglWingGetWettedArea', wingUID)

rainman110, 2019-08-16:
What problem do you get?

yaoliu19, 2019-08-19:
wingUID = tigl_h.wingGetUID(1)
    ref_area = 2*tigl_h.wingGetReferenceArea(1,1)
    surf_area = tigl_h.wingGetSurfaceArea(1)
    span = tigl_h.wingGetSpan(wingUID)
    wetted_area = tigl_h.wingGetWettedArea(wingUID)   

  File "C:\Users\tiglwrapper.py", line 2771, in wingGetWettedArea
    errorCode = self.lib.tiglWingGetWettedArea(self._handle, _c_wingUID, ctypes.byref(_c_wettedAreaPtr))

OSError: exception: access violation reading 0x0000000000000000

yaoliu19, 2019-08-19:
wingUID = tigl_h.wingGetUID(1)
ref_area = 2*tigl_h.wingGetReferenceArea(1,1)
surf_area = tigl_h.wingGetSurfaceArea(1)
span = tigl_h.wingGetSpan(wingUID) # ok
wetted_area = tigl_h.wingGetWettedArea(wingUID)   # error

File "C:\Users\tiglwrapper.py", line 2771, in wingGetWettedArea
errorCode = self.lib.tiglWingGetWettedArea(self._handle, _c_wingUID, ctypes.byref(_c_wettedAreaPtr))

OSError: exception: access violation reading 0x0000000000000000

rainman110, 2019-08-19:
Can you please provide the cpacs file? Else, it will be hard to reproduce the error.

yaoliu19, 2019-08-19:
<cpacs>
    <header>
        <name>genericSystem example file</name>
        <description>Example file for use of genericSystem components</description>
        <creator>Jonas Jepsen from TU-Hamburg-Harburg</creator>
        <timestamp>2017-05-31T08:55:23</timestamp>
        <version>1.0</version>
        <cpacsVersion>3.0</cpacsVersion>
    </header>
    <vehicles>
        <aircraft>
            <model uID="Aircraft1">
                <name>genericSystem example</name>
                <description>Container to show use of genericSystem nodes</description>
				<systems>
					<genericSystems>
						<genericSystem uID="kjashdi">
							<geometricBaseType>sphere</geometricBaseType>
							<transformation>
								<scaling uID="Aircraft1_Fuselage1_Transf_Sca">
									<x>1.0</x>
									<y>1.0</y>
									<z>1.0</z>
								</scaling>
								<rotation uID="Aircraft1_Fuselage1_Transf_Rot">
									<x>0.0</x>
									<y>0.0</y>
									<z>0.0</z>
								</rotation>
								<translation refType="absGlobal" uID="Aircraft1_Fuselage1_Transf_Transl">
									<x>0.0</x>
									<y>0.0</y>
									<z>0.0</z>
								</translation>
							</transformation>
						</genericSystem>
						<genericSystem uID="dfhert">
							<geometricBaseType>cube</geometricBaseType>
							<transformation>
								<scaling uID="Aircraft1_Fuselage1_Transf_Sca">
									<x>1.0</x>
									<y>1.0</y>
									<z>1.0</z>
								</scaling>
								<rotation uID="Aircraft1_Fuselage1_Transf_Rot">
									<x>0.0</x>
									<y>0.0</y>
									<z>0.0</z>
								</rotation>
								<translation refType="absGlobal" uID="Aircraft1_Fuselage1_Transf_Transl">
									<x>2.0</x>
									<y>0.0</y>
									<z>0.0</z>
								</translation>
							</transformation>
						</genericSystem>
						<genericSystem uID="erztghf">
							<geometricBaseType>cylinder</geometricBaseType>
							<transformation>
								<scaling uID="Aircraft1_Fuselage1_Transf_Sca">
									<x>1.0</x>
									<y>1.0</y>
									<z>1.0</z>
								</scaling>
								<rotation uID="Aircraft1_Fuselage1_Transf_Rot">
									<x>0.0</x>
									<y>0.0</y>
									<z>0.0</z>
								</rotation>
								<translation refType="absGlobal" uID="Aircraft1_Fuselage1_Transf_Transl">
[... 162 lines cut ...]
                                <name>Segment Name</name>
                                <description>Segment Description</description>
                                <fromElementUID isLink="True">Aircraft1_Wing1_Sec1_Elem1</fromElementUID>
                                <toElementUID isLink="True">Aircraft1_Wing1_Sec2_Elem1</toElementUID>
                            </segment>
                        </segments>
                    </wing>
                </wings>
            </model>
        </aircraft>
        <profiles>
            <wingAirfoils>
                <wingAirfoil uID="NACA0009">
                    <name>NACA0009 Airfoil</name>
                    <description>NACA 4-digit symmetrical airfoil with a thickness of 9percent</description>
                    <pointList>
                        <x mapType="vector">1.0;0.95;0.9;0.8;0.7;0.6;0.5;0.4;0.3;0.25;0.2;0.15;0.1;0.075;0.05;0.025;0.0125;0.005;0.0;0.005;0.0125;0.025;0.05;0.075;0.1;0.15;0.2;0.25;0.3;0.4;0.5;0.6;0.7;0.8;0.9;0.95;1.0</x>
                        <y mapType="vector">0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0</y>
                        <z mapType="vector">0.00095;0.00605;0.01086;0.01967;0.02748;0.03423;0.03971;0.04352;0.04501;0.04456;0.04303;0.04009;0.03512;0.0315;0.02666;0.01961;0.0142;0.0089;0.0;-0.0089;-0.0142;-0.01961;-0.02666;-0.0315;-0.03512;-0.04009;-0.04303;-0.04456;-0.04501;-0.04352;-0.03971;-0.03423;-0.02748;-0.01967;-0.01086;-0.00605;-0.00095</z>
                    </pointList>
                </wingAirfoil>
            </wingAirfoils>
        </profiles>
    </vehicles>
</cpacs>

yaoliu19, 2019-08-19:
genericSystemShapes.xml, from cpacs website, thanks

yaoliu19, 2019-08-20:
[genericSystemShapes.zip](https://github.com/DLR-SC/tigl/files/3520188/genericSystemShapes.zip)

joergbrech, 2019-08-20:
This actually is a bug: TiGL uses the parent of the wing to calculated the wetted area and crashes, if the wing has no parent, i.e. a fuselage. 

Thanks for pointing this out! I will take care of it!

PR #618 (2019-08-22): bugfix in tiglWingGetWettedArea
fixes #614

## Description

 - To calculate the wetted area of the wing, the wing must be cut with the fuselage, which is read as the parent geometric component of the wing. But the parentUID element of the wing is optional, and TiGL used to crash if no parent were to be found.

 - Now, the API function tiglGetWettedArea returns the wetted area of the wing if the wing has a parent. Else, it returns the surface area of the wing (API change).

 - In addition, a minor bug in line 237 of CTiglUidManager.cpp was fixed.

## How Has This Been Tested?

A unit test has b [... cut]
commit 8ae5d98f (2019-08-22): bugfix in tiglWingGetWettedArea

fixes #614
closed 2019-08-22 by joergbrech