DLR-SC/tigl #615: Compilation fails with VS013

spirritus, 2019-08-05:
In the Wiki you reommend VisuaStudio 2010. However you use newer capabilities which where introduced with VS2015.

Namely:
UniquePtr.h(25) uses [[deprecated]] - this does not compilie with VS2013
Cache.h(39) uses defaulted copy constructor - this dows not compile with VS2013.

.....\tigl\src\Cache.h(39): error C2610: 'tigl::ScopeLockedReference<T>::ScopeLockedReference(tigl::ScopeLockedReference<T> &&)' : is not a special member function which can be defaulted (C:\data\gitRepos\tigl\src\CCPACSXsiIsoLine.cpp)
1>          c:\data\gitrepos\tigl\src\Cache.h(47) : see reference to class template instantiation 'tigl::ScopeLockedReference<T>' being compiled

The deprecated flag can be removed to compile.
However, I am a bit at a loss how to convert the "default" to what should happen in the constructor.

spirritus, 2019-08-05:
In Cache.h, replaced ScopeLockedReference(ScopeLockedReference&&) = default; with

`		ScopeLockedReference(ScopeLockedReference&& ref) :
			m_reference(std::move(ref.m_reference)),
			m_mutex(std::move(ref.m_mutex))
		{}`

No clue if this is correct. Now other errors appear:

`......\tigl\src\.\generated/TixiHelper.h(53): error C2144: syntax error : 'auto' should be preceded by ';' (......\tigl\src\configuration\CCPACSConfiguration.cpp)
......\tigl\src\.\generated/TixiHelper.h(53): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int ((......\tigl\src\configuration\CCPACSConfiguration.cpp)`

joergbrech, 2019-08-05:
Interesting. Have you tried compiling with VS2015? TiGL now uses some C++11 features that were not used before and as fas as I know, VS2013 does not have complete C++11 supprt. 

https://dlr-sc.github.io/tigl/embracing-new-technology-abandon-python-2-support-and-pre-c11-compilers.html

PS: I will update the outdated entry in our Wiki, sorry for the confusion

spirritus, 2019-08-05:
I could get a hold of Visual Studio 2019 Community. Here cmake does not generate the compiler flags. But I could reuse the ones from VS2013.

Also, for the java bindings I use openjdk 11. For that to compile I had to copy some files from cmake install dir to the make dir in the project, namely:
`FindJava.cmake
CMakeFindJavaCommon.cmake
FindPackageHandleStandardArgs.cmake
FindPackagemessages.cmake`
Maybe update your cmake?

Back to the compile problem. I reverted my change. Now with VS2019 the error is:
`2>......\tigl\src\math\CTiglProjectPointOnCurveAtAngle.cpp(228,40): error C2664:  "math_BFGS::math_BFGS(const math_BFGS &)" : Konvertierung von Argument 1 von "`anonymous-namespace'::ProjectObjective" in "const Standard_Integer" nicht möglich`

Also I am not to conviced line 193 in CTiglProjectPointOnCurveAtAngle.cpp is correct:
`void CTiglProjectPointOnCurveAtAngle::CTiglProjectPointOnCurveAtAngle::Compute() const`

rainman110, 2019-08-05:
You are right. Line 191 is wrong. Strange, that it compiles with clang, gcc and msvc 2015.

rainman110, 2019-08-05:
@spirritus Which opencascade version are you using?

spirritus, 2019-08-06:
That would be 7.3.0

rainman110, 2019-08-07:
Indeed, the API of math_BFGS has changed slightly at 6.9.1. We are using 6.8.0. this could be easily adapted however.

PR #622 (2019-09-24): math_BFGS call changed since OCC version 6.9.1
fixes #615
commit 652a9a28 (2019-09-24): math_BFGS call changed since OCC version 6.9.1

fixes #615
closed 2019-09-24 by rainman110